gloomyandy/RepRapFirmware #14: SKR 1.3 How do you configure BLTouch in LPCConfigurator?

Crunch69, 2020-06-26:
SKR 1.3 How do you configure BLTouch in LPCConfigurator?

Z Probe pin assignment Input pin selection is there z stop 

What are the correct selections for the Modulation Pin and  PWM Pin?

gloomyandy, 2020-06-28:
If you need help please use either the RepRap forum or the discord server.

closed 2020-06-28 by gloomyandy

eightheads, 2020-07-15:
> SKR 1.3 How do you configure BLTouch in LPCConfigurator?
> 
> Z Probe pin assignment Input pin selection is there z stop
> 
> What are the correct selections for the Modulation Pin and PWM Pin?

Here is my working configuration for my SKR v1.4. The configurator does not allow for the proper options, you will need to edit your config file yourself.
`; Z-Probe

M950 S0 C"servo0"                              ; create servo pin 0 for BLTouch

M558 P9 C"^zstop" H5 F120 T6000                ; set Z probe type to bltouch and the dive height + speeds

G31 P500 X-35 Y-5 Z0.6                         ; set Z probe trigger value, offset and trigger height

M557 X15:215 Y15:195 S40                       ; define mesh grid`