gloomyandy/RepRapFirmware #15: LPCConfigurator will not allow adding Fans on I/O mapping or Fan page

Crunch69, 2020-06-27:
LPCConfigurator will not allow adding Fans on I/O mapping or Fan page.  On the LPC Configurator I/O page pressing the green plus button does not allow the addition of a fan the red negative will delete

eightheads, 2020-07-11:
I noticed tonight the stepper sections labels are in the wrong positions
[image]

As compared to the Duet configurator
[image]

jaysuk, 2020-07-11:
Thanks eightheads for pointing that out.
It has now been fixed

closed 2020-07-11 by jaysuk