gloomyandy/RepRapFirmware #18: Bigtreetech SKR v1.4 BLTouch

cobramike, 2020-07-15:
So i followd your new instrucktion on BLTouch. But its not trigering. Also when i unplug the probe its triggerd.

Heres my config

; Configuration file for SKR v1.4 (firmware version 3)
; executed by the firmware on start-up
;
; generated by RepRapFirmware Configuration Tool v3.1.3-LPC on Sat Jul 11 2020 16:57:13 GMT+0200 (Midden-Europese zomertijd)

; General preferences
G90                                            ; send absolute coordinates...
M83                                            ; ...but relative extruder moves
M550 P"Artillery genius"                       ; set printer name

; Network
M552 S1
M552 P0.0.0.0 S1                               ; enable network and acquire dynamic address via DHCP
M586 P0 S1                                     ; enable HTTP
M586 P1 S0                                     ; disable FTP
M586 P2 S0                                     ; disable Telnet

; Drives
M569 P0 S0 T0.1:0.1:0.02:0.02                  ; physical drive 0 goes backwards using TMC220x driver timings
M569 P1 S0 T0.1:0.1:0.02:0.02                  ; physical drive 1 goes backwards using TMC220x driver timings
M569 P2 S1 T0.1:0.1:0.02:0.02                  ; physical drive 2 goes forwards using TMC220x driver timings
M569 P3 S0 T0.1:0.1:0.02:0.02                  ; physical drive 3 goes backwards using TMC220x driver timings
M584 X0 Y1 Z2 E3                               ; set drive mapping
M350 X16 Y16 Z16 E16 I1                        ; configure microstepping with interpolation
M92 X80.12 Y80.12 Z399.78 E445.00              ; set steps per mm
M566 X900.00 Y900.00 Z12.00 E120.00            ; set maximum instantaneous speed changes (mm/min)
M203 X6000.00 Y6000.00 Z180.00 E1200.00        ; set maximum speeds (mm/min)
M201 X500.00 Y500.00 Z20.00 E250.00            ; set accelerations (mm/s^2)
M906 X800 Y800 Z1200 E800 I30                   ; set motor currents (mA) and motor idle factor in per cent
M84 S30                                        ; Set idle timeout

; Axis Limits
M208 X0 Y0 Z0 S1                               ; set axis minima
M208 X225 Y225 Z250 S0                         ; set axis maxima

; Endstops
M574 X1 S1 P"!xstop"                            ; configure active-high endstop for low end on X via pin xstop
M574 Y1 S1 P"!ystop"                            ; configure active-high endstop for low end on Y via pin ystop
M574 Z1 S2                                      ; configure probe port for BLtouch output
 
; Z-Probe
M558 P9 H6 F120 T8000 X0 Y0 Z1 C"^probe"       ; disable Z probe but set dive height, probe speed and travel speed              ; set Z probe type to bltouch, set dive height, probe speed + travel speed
M950 S0 C"servo0"                              ; create servo pin 0 for BLTouch
G31 P500 X-31 Y80 Z3.25                        ; set Z probe trigger value, offset and trigger height (smaller number means nozzle further from bed)
M557 X35:275 Y35:300 S40                       ; define mesh grid

; Heaters
M308 S0 P"bedtemp" Y"thermistor" T100000 B4092 ; configure sensor 0 as thermistor on pin bedtemp
M950 H0 C"bed" T0                              ; create bed heater output on bed and map it to sensor 0
M307 H0 B0 S1.00                               ; disable bang-bang mode for the bed heater and set PWM limit
M140 H0                                        ; map heated bed to heater 0
M143 H0 S120                                   ; set temperature limit for heater 0 to 120C
M308 S1 P"e0temp" Y"thermistor" T100000 B4092  ; configure sensor 1 as thermistor on pin e0temp
M950 H1 C"e0heat" T1                           ; create nozzle heater output on e0heat and map it to sensor 1
M307 H1 B0 S1.00                               ; disable bang-bang mode for heater  and set PWM limit

; Fans
M950 F0 C"fan0" Q500                           ; create fan 0 on pin fan0 and set its frequency
M106 P0 S0 H-1                                 ; set fan 0 value. Thermostatic control is turned off
M950 F1 C"he1" Q500                            ; create fan 1 on pin he1 and set its frequency
M106 P1 H1 T50                                 ; set fan 1 value. Thermostatic control is turned on at 50 deg C
 
; Tools
M563 P0 D0 H1 F0                               ; define tool 0
G10 P0 X0 Y0 Z0                                ; set tool 0 axis offsets
G10 P0 R0 S0                                   ; set initial tool 0 active and standby temperatures to 0C
 
; Miscellaneous
M575 P1 S2 B57600                              ; enable support for PanelDue
M501                                           ; load saved parameters from non-volatile memory
T0                                             ; select first tool

jonaskonig, 2020-07-15:
Did you change the homez and homeall config?

jaysuk, 2020-07-15:
Did you also check with the macro's that it triggers?

closed 2020-07-15 by cobramike