gloomyandy/RepRapFirmware #19: Dual Z axis creating an U drive separately which is so not homing

kiki0000, 2020-07-20:
Hi. I have an SKR 1.4 configured with the great Reprap firmware.
Similarly to my Duet wifi I configured a dual Z axis to have independent Z axis compensation with a BLtouch.
I have assigned it the following way: M584 X0 Y1 Z2:4 U4 E3   
My problem is that it is showing up in the Duet web control as a separate U drive and so due to the reason it is not homing it always drops an error that it is not homed and a G29 mesh bed leveling cannot be initialised.
Am I doing something wrong?

Thank you for your support in advance.

evgen2910, 2020-07-21:
you must write **M584 X0 Y1 Z2:4 E3** - this means that you have two Z motor connected to Z and E1 driver. In board.txt must be set **stepper.numSmartDrivers = 5;** .
If you want independent Z-axis compensation, you must write to config.g  **M671 X-45.0:355.0 Y150.0:150.0 S0.5   ; Z leadscrews are at (-45,150), (150,355)** with you location motor & put `M291 P"Probing mesh grid process started" R"Probing.." S1 T2
M561                                       ; clear any bed transform
M671 S5                                    ; specify a maximum correction per leadscrew a 5mm
G28                                        ; home
M401                                       ; deploy Z probe (omit if using bltouch)
G30 P0 X10 Y150 Z-99999                    ; probe near a leadscrew, half way along Y axis
G30 P1 X300 Y150 Z-99999 S2                ; probe near a leadscrew and calibrate 2 motors 1pass
G30 P0 X10 Y150 Z-99999                    ; probe near a leadscrew, half way along Y axis
G30 P1 X300 Y150 Z-99999 S2                ; probe near a leadscrew and calibrate 2 motors 2 pass
G30 P0 X10 Y150 Z-99999                    ; probe near a leadscrew, half way along Y axis
G30 P1 X300 Y150 Z-99999 S2                ; probe near a leadscrew and calibrate 2 motors 3 pass
M561               		                   ; clear any bed transform
G28				                           ; home all axis bb` 
to your bed.g whis your position probe at X and Y

kiki0000, 2020-07-21:
evgen2910 many thanks for your help! My only issue was that I followed the Duet way and I also added the U4 after the two Z mapping. I only had to remove this and everything works perfectly.

closed 2020-07-21 by kiki0000