gloomyandy/RepRapFirmware #21: SKR 1.3 Bed Heater Constantly Faulting

JW97, 2020-07-24:
Hi,

I'm running 3.1.1-14 on an SKR 1.3 in SBC mode and am having issues the bed heater faulting out every time I try to heat it.

I've ran autotune starting from room temp numerous times, and I've even tried manually increasing the dead time but I'm still getting 'heater fault, temperature rising more slowly than the expected 1.6deg/s' on the bed heater.

Given that it heats up fine when auto-tuning but faults under normal operation I wanted to raise the issue in case it was a firmware problem?

I've manually checked the resistance of the bed and all looks within spec.

jonaskonig, 2020-07-24:
Hey,
had the same issue. In duet firmware, thermal runaway protection is not active while tuning. 
I had to put the values in the configuration file to get them saved. Here is a link for the gcode:
https://duet3d.dozuki.com/Wiki/Gcode#Section_M304_Set_PID_parameters_Bed

JW97, 2020-07-24:
I have tried defining my tuning parameters in both config-override.g & config.g but neither seems to be addressing the issue.
Any ideas?

jaysuk, 2020-07-24:
What command are you running to PID tune the bed?
What command are you running to save the PID result?
If you reboot your machine and run M307 H0, are the outputted values the values that you get from the PID tune?

JW97, 2020-07-24:
M303 H0 S60 to autotune
M500 after to save to config-override.txt
8GB SD Card I believe (64GB in the Pi)

However, they aren't showing up if I run M307 H0. I'll add these to the end of config.g and see if they load once this prints finished (I had to clear the fault 6 times to get to operating bed temp).

JW97, 2020-07-24:
I think that's sorted it for now. Many Thanks!

closed 2020-07-24 by JW97