gloomyandy/RepRapFirmware #23: Max31855

overdrive2max, 2020-08-03:
Hi
Is it possible to use a MAX31855 termocouple amplifier with this port of the RRF?

jaysuk, 2020-08-04:
yes, it should be possible. We've had users report success before

overdrive2max, 2020-08-04:
Thank you for your response. 
My wifi disconnects as soon as I call:
M308 S1 P"P3.26" Y"thermocouple-max31855"
CLK connected to 0.15 and SO connected to 0.17
Do I need to specify anything extra?

jaysuk, 2020-08-04:
board.txt
//Soft SPI Temp sensor to use with MAX31865+PT100
softwareSPI.pins = {1.18,1.19,1.20}
heat.spiTempSensorCSPins = {1.21, NoPin}
heat.spiTempSensorChannel = 2

Gcode
M308 S1 P"P1.21" Y"rtd-max31865"

overdrive2max, 2020-08-04:
Thank you. I will try that shortly.

overdrive2max, 2020-08-04:
It is working. Thank you very much for the assistance.

closed 2020-08-04 by jaysuk