gloomyandy/RepRapFirmware #25: SKR V1.4 Turbo+ESP8266 "Failed to initialise WiFi module, code -7"

gitxmax, 2020-08-08:
Having an issue, with activating ESP8266 wifi and Skr V1.4 Turbo
Spend hours trying to troubleshoot the error, rechecked all the connections, resistor values and config files, but always getting the error.

`Printer is now online.
>>> M552 S0
SENDING:M552 S0
WiFi module started
ok T0:-273.1 /0.0 B:-273.1 /0.0
Error: Failed to initialise WiFi module, code -7`

M587 S"your SSID" P"your password" results in"
`Error: M587: Failed to add SSID to remembered list`

When I send "M552 S0", ESP8266  blue led flashes briefly.
Tried re-flashin the module multiple times, with the latest "DuetWifiServer.bin". Always success.

Skr V1.4 Turbo  is flashed with the latest firmware-wifi.bin (renamed to firmware.bin)

[board.txt](https://github.com/gloomyandy/RepRapFirmware/files/5046176/board.txt)

P-C-R, 2020-08-08:
Sounds like a hardware issue. Not Software!

gitxmax, 2020-08-08:
It was wrong wiring ...CS/MOSI/MISO/SCLK not as default, labeled on ESP8266.

closed 2020-08-08 by gitxmax

gloomyandy, 2020-08-09:
The problem was hardware configuration, now resolved. Closing.