gloomyandy/RepRapFirmware #26: problems with RRF COnfig tool for LPC an BLTouch

braini1511, 2020-08-14:
Sorry for bothring but is there any know issue with the SKR1.4 and a configured BL-Touch connection on the I/O mapping page of the RRF Config Tool (LPC-Port)?

When i configure any Pins for the BL-Touch there i get an error on the last Page saying that there are errors on the I/O-mapping page. When i leave the BLtouch unconnected anything is fine i can get the config tool to spit out the disired files.

Best regards
René

evgen2910, 2020-08-14:
LPC configuration tool may trable whith configurating BlTouch... find any example & use it to configuring bltouch in config.g

evgen2910, 2020-08-14:
You may use this - https://github.com/jaysuk/LPCRRFUserConfigs/tree/master/SecKit-SK-GO%20-%20SKR1.3%20-%20ESP%20-%20L3LK3K

closed 2020-08-17 by jaysuk