gloomyandy/RepRapFirmware #30: Enable PSU_CONTROL

wolfgangmauer, 2020-08-24:
I use a BTT Relay connected to pin 1.24 on my SKR 1.3

In Marlin i have defined:
#define PSU_CONTROL
#define PSU_NAME "Power Supply"
#define PS_ON_PIN P1_24
#define PSU_ACTIVE_HIGH true

Is this possible in RRF?
Thanks for Help!

P-C-R, 2020-08-25:
Yep it is. In the Discord someone posted the solution. Think it will be added shortly here

P-C-R, 2020-08-26:
Add this to your board.txt

`atx.powerPin = 2.00 ; connected to the servo pin.
atx.powerPinInverted = 0;`

and this at the end of your config.g

`M80`

Then you can add the M81 command when you want to power it off:
[https://duet3d.dozuki.com/Wiki/Gcode#Section_M81_ATX_Power_Off](M81)

i added pictures. They are not mine!

[image]
[image]

wolfgangmauer, 2020-08-26:
I've tried the following(with M80 in config.g):

atx.powerPin = 1.24 ; connected to the pin.
atx.powerPinInverted = 0;

and

atx.powerPin = 1.24 ; connected to the pin.
atx.powerPinInverted = 1;

both with same result.
After a few seconds the relay switches off.
I hope it doesn't only work with the servo pin, my BLTouch is on it ;-)

P-C-R, 2020-08-26:
Dear Wolfgang,

can you post a picture of the wiring and the output of M122 P200?

P-C-R, 2020-08-26:
@wolfgangmauer 

https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/issues/5#issuecomment-577595738

try this. They have the same problem in the thread

wolfgangmauer, 2020-08-26:
Hmmm, with my wiring and the configuration above(Marlin) everything works fine.
I will try the TH1(0.25) Pin

[image]
[image]

P-C-R, 2020-08-26:
Are you supplying 5v to the relay? On the pictures there is no cable at the
VIN

wolfgangmauer, 2020-08-26:
No I did not. As described by BTT, this is only required if you want to detect a 5 V short circuit.

 

Even with pin 0.25 (TH1) it doesn't work ...

 

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

wolfgangmauer, 2020-08-26:
I'm using this simple connection, Black/Red connected to Z-MAX on my SKR 1.3

[image]

P-C-R, 2020-08-26:
What you could try is this thing

P-C-R, 2020-08-26:
You could try switching the relay with the M42 command

P-C-R, 2020-08-26:
https://duet3d.dozuki.com/Wiki/Gcode#Section_M42_Switch_I_O_pin

M950 P4 C"1.24"
M42 P4 S0 in the config.g

And then switching IT via M42 P4 S1 in the command line

wolfgangmauer, 2020-08-26:
I tried C"1.24" C"!1.24" C"^1.24" C"^!1.24"
Nothing happens :-(

Since I am not very familiar with RRF there may be something wrong with my 
[config.g](https://github.com/gloomyandy/RepRapFirmware/files/5131400/config.g.txt)

wolfgangmauer, 2020-08-27:
I think BTT designed the module as a "power off after print", nothing else.
Not to turn the printer on and off. This would also require an additional power source(5V).
An M80 at the end of the config.g should not be necessary.
In the end-gcode of the slicer an M81(as a final order) and everything is fine.

P-C-R, 2020-08-27:
So works now?

Best

wolfgangmauer, 2020-08-27:
No it doesn't work.
I have to print something overnight every now and then that's why I'm back on Marlin.
But can try something if there is something new ...

wolfgangmauer, 2020-08-28:
I finally found a working solution :-) SKR 1.3 / BTT Relay 1.2
Only Reset is connected to a switch(only works to turn on power) and the z-max as i shown in my previous picture!

board.txt
atx.powerPin = 1.24 ; connected to the pin Z-MAX.(Signal/GRD)
atx.powerPinInverted = 1;

Removing all M80 from any configuration!
Now if you try M81 nothing happens, but with M80 the power shut down! Yeahhh
I don't know now if M80/M81 is swapped in RRF_LPC, but this is how it works ...

wolfgangmauer, 2020-08-31:
I looked at the Marlin-Source, it works there.
These places would have to be adjusted?! @gloomyandy 

`
bool Platform::AtxPower() const noexcept
{
    **return IoPort::ReadPin(ATX_POWER_PIN);**
}

void Platform::AtxPowerOn() noexcept
{
	deferredPowerDown = false;
#ifdef __LPC17xx__
    **IoPort::WriteDigital(ATX_POWER_PIN, ATX_POWER_INVERTED);**
#else
    IoPort::WriteDigital(ATX_POWER_PIN, true);
#endif
}

void Platform::AtxPowerOff(bool defer) noexcept
{
	deferredPowerDown = defer;
	if (!defer)
	{
#if HAS_MASS_STORAGE
		if (logger != nullptr)
		{
			logger->LogMessage(realTime, "Power off commanded");
			logger->Flush(true);
			// We don't call logger->Stop() here because we don't know whether turning off the power will work
		}
#endif
#ifdef __LPC17xx__
		**IoPort::WriteDigital(ATX_POWER_PIN, !ATX_POWER_INVERTED);**
#else
		IoPort::WriteDigital(ATX_POWER_PIN, false);
#endif
	}
}
`

wolfgangmauer, 2020-08-31:
Verified with self compiled binary

McPaul0, 2020-10-07:
> I finally found a working solution :-) SKR 1.3 / BTT Relay 1.2
> Only Reset is connected to a switch(only works to turn on power) and the z-max as i shown in my previous picture!
>
> board.txt [...]

No, M80 81 are not swapped.
Config.txt
atx.powerPinInverted = 0
Config.g
add M80 in General preferences to turn the relay on.
So now, M81 will work as it should be, with very useful S1 feature.

wolfgangmauer, 2020-10-07:
@McPaul0 So you have a SKR1.3 and using "atx.powerPinInverted = 0" with a BTT Relay 1.2 ???

McPaul0, 2020-10-08:
SKR 1.4 turbo, relay is my own.
But BTT Relay 1.2 and MKS PWC (exept for Lerdge) needs HIGH on input to turn ON, and LOW to turn OFF. So no need in pin inverting.
I've used Pin 2.00 for servo

Minims, 2020-10-22:
Trying to make the same on SKR Pro, but does not seems to fully working. M80 stop the printer.
atx.powerPinInverted has no action. 0 or 1 do the same thing.

I have add this to board.txt :
```
// PS ON
atx.powerPin = C.9
atx.powerPinInverted = 0;
```

c1000mg, 2020-12-14:
Has anyone try the "Makerbase MKS PWC Power Monitoring auto power off" ?

[image]

closed 2021-01-21 by gloomyandy