gloomyandy/RepRapFirmware #31: WiFi connector "dummy" ?

Drzet, 2020-08-24:
After trying unsuccessfully to make ESP2866 module work I realised that the pins mentioned in config : 0.28, 1.30 and 1.31 are located over in the expansion section. The middle pins of 8pin WiFi connector, according to official pinout diagram are "N/C" which as I understand stands for "not connected" ?
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4PIN.pdf

Please correct me if I wrong. Otherwise, is there a way to reassign these pins to utilise WiFi connector?

jaysuk, 2020-08-24:
No, there isn't.
The only way that works is to follow the instructions on the wiki

jaysuk, 2020-08-24:
N

closed 2020-08-24 by jaysuk