gloomyandy/RepRapFirmware #33: [FR] Shutdown SBC after printjob finish

wolfgangmauer, 2020-08-28:
In combination with a Relay it would be nice if it is possible to shutdown the SBC after print is finish, before power off

P-C-R, 2020-08-28:
@gloomyandy is doing something like this!

P-C-R, 2020-08-29:
You could make this on the pi

P-C-R, 2020-08-29:
Or @wolfgangmauer

P-C-R, 2020-08-29:
https://github.com/wilriker/execonmcode

wolfgangmauer, 2020-08-29:
panic: dial unix /var/run/duet.sock: connect: no such file or directory :-(

P-C-R, 2020-08-29:
Will test it this weekend!

P-C-R, 2020-08-29:
Tested it with my SKR GTR and the SBC. Worked!
I only got your first error message when the PI was not connected to the SKR! Will try later with a SKR 1.3

P-C-R, 2020-08-29:
The M7722 Error will come Always. Because its no rrf command. But the Pi will See it

wolfgangmauer, 2020-08-29:
I think the installed golang version is too old

P-C-R, 2020-08-29:
? have to testetd it with the PI and SKR connected? I downloaded the armv7 version from here
https://github.com/wilriker/execonmcode/releases/tag/v5.1

wolfgangmauer, 2020-08-29:
Upps, sorry,
I will try....

P-C-R, 2020-08-29:
I will later make an Tutorial!

Best

wolfgangmauer, 2020-08-29:
Would be very great!

wolfgangmauer, 2020-08-29:
With the "correct" binary and service, M7722 works

P-C-R, 2020-08-29:
nice to hear!

P-C-R, 2020-08-29:
for making a service out of it see:
https://www.raspberrypi.org/documentation/linux/usage/systemd.md

wolfgangmauer, 2020-08-29:
Is there something like a "jobfinish.g" or put it in slicer instead?

P-C-R, 2020-08-29:
a sample file is in the repo from execonmcode

P-C-R, 2020-08-29:
You can Put IT in the slicer or in a macro. You can call every macro with
M98 P"macroname.g"

wolfgangmauer, 2020-08-29:
I think I choose the slicer, the shutdown is triggered mostly at night when print is finished

P-C-R, 2020-08-29:
Yep but Oh could write in your endcode of the slicer the M98 command.

And then create the macro at dwc with the ATX and SBC shutdown command ;)

P-C-R, 2020-08-29:
For Future reference

https://youtu.be/WxcLDUF9zMs

And
https://youtu.be/q2Wl0Ew97PU

wolfgangmauer, 2020-08-31:
Works perfect, close this?

closed 2020-09-01 by gloomyandy

CrazyCreator13, 2021-05-18:
hello ... I think the script is running fine on my pi with connected duet 3 ... but M7722 give me the popup 

Warning: M7722: Command is not supported

and the pi is running always