gloomyandy/RepRapFirmware #35: [BUG] Filament runout not working

wolfgangmauer, 2020-08-30:
I have defined a filament runout sensor, the message appears, but the printhead do not move to the position set in pause.g
It looks like pause.g is not called

;config.g
; Filament sensor Extruder 0
M591 P2 D0 C"xstopmax" S1

and my
 
; pause.g
; called when a print from SD card is paused
;
; generated by RepRapFirmware Configuration Tool v3.1.3-LPC on Mon Aug 03 2020 18:49:08 GMT+0200 (Mitteleuropäische Sommerzeit)
M83            ; relative extruder moves
G1 E-10 F3600  ; retract 10mm of filament
G91            ; relative positioning
G1 Z5 F360     ; lift Z by 5mm
G90            ; absolute positioning
G1 X0 Y0 F6000 ; go to X=0 Y=0

P-C-R, 2020-08-30:
Are you printing from SBC?

P-C-R, 2020-08-30:
If you call M98 P"pause.g" it works?

wolfgangmauer, 2020-08-30:
Yes

 

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

P-C-R, 2020-08-30:
You can try this

M950 J1 C"xstopmax" ; Define Input J1 for pin duex.e2stop M581 P1 S1 T2 C0 ; Connect Input J1 (P1) to trigger 2 (T2) always (C0) for inactive to active (S1 - also default)

wolfgangmauer, 2020-08-30:
Yes, if i call it it works fine

 

https://www.avast.com/antivirus

P-C-R, 2020-08-30:
You have to create a file called trigger2.g

wolfgangmauer, 2020-08-30:
Yes for testing with M591 ... T2

P-C-R, 2020-08-30:
So how is it working?;)

wolfgangmauer, 2020-08-30:
Not at all!
Without T2 Dialog popup but stays on place, no retraction, no move.
With T2 the trigger.g is executed, moves, retracts, until Dialog popup, hit Ok -> loop forever

P-C-R, 2020-08-30:
Have you tried the M581 method or did you add T2 to M591

wolfgangmauer, 2020-08-30:
M591 P2 D0 C"xstopmax" S1 T2

wolfgangmauer, 2020-08-30:
So M581 is "better" ?

P-C-R, 2020-08-30:
Test if its working or not

wolfgangmauer, 2020-08-30:
So should be
M581 P"xstopmax" T2 S1 
?

P-C-R, 2020-08-30:
No
M950 J1 C"xstopmax" ; Define Input J1 for pin duex.e2stop M581 P1 S1 T2 C0 ; Connect Input J1 (P1) to trigger 2 (T2) always (C0) for inactive to active (S1 - also default)

wolfgangmauer, 2020-08-30:
M950 J1 C"xstopmax"
I can't start a print job anymore :-(

P-C-R, 2020-08-30:
OK because its in the a false state

P-C-R, 2020-08-30:
Try inverting the M950 Like

M950 J1 C"!xstopmax"

wolfgangmauer, 2020-08-30:
I have to try again, i think skr is crashed

P-C-R, 2020-08-30:
Then manually delte the Line from config.g If the skr does Not restart.

wolfgangmauer, 2020-08-30:
In this case Marlin is very easy ;-)

P-C-R, 2020-08-30:
Could be a Bug in the SBC Version. @gloomyandy

wolfgangmauer, 2020-08-30:
With

P-C-R, 2020-08-30:
With M950 and M581? You need both

wolfgangmauer, 2020-08-30:
Ohhh, I'm a little confused ...

Sould be 

M581 P"xstopmax" T2 S1
M950 J1 C"xstopmax"

P-C-R, 2020-08-30:
No

M950 J1 C"xstopmax"
M581 P1 S1 T2 C0

Or

M950 J1 C"!xstopmax"
M581 P1 S1 T2 C0

wolfgangmauer, 2020-08-30:
M950 J1 C"xstopmax"
M581 P1 S1 T2 C0
Now calls the trigger2.g.
Is it nessesary to stop the trigger(M581 S-1 T2) inside trigger2.g ?
An at the end restart with 
M581 P1 S1 T2
M582 T2
?

wolfgangmauer, 2020-08-30:
Same behavier as in my first post.
Everything is executed until the M291 Line, after push ok nothing happens(I only use the Duet Web Control)

;0:/sys/trigger2.g
M25
M400
G4 P200              ; Wait 200 milliseconds
M83                  ; relative extruder moves
G1 E10 F500          ; Extrude 2mm to warm the heatbreak filament
G4 P500              ; Wait 500 milliseconds
G1 E-10 F500
G1 E-670 F3600       ; retract 670mm of filament
M400                 ; Wait for current moves to finish
M291 P"Insert Filament" R"Filament runout detected!" S2
M83                  ; relative extruder moves
G1 E670 F3600        ; extrude 670mm of filament
G1 E10 F500
M400                 ; Wait for current moves to finish
M24

Maybe this is the problem 
M292 -> This command is sent by the user interface when the user acknowledges a message that was displayed because of a M291 command with parameter S=2 or S=3.

P-C-R, 2020-08-31:
Will Look into it when i have time!

P-C-R, 2020-08-31:
Tested it on my GTR with WiFi build. For me it works.

P-C-R, 2020-08-31:
M591 P1 C"PH_6" S1 D0

With the Standard pause.g

Will Test it with SBC sometime later

wolfgangmauer, 2020-08-31:
Yes, please don't simplify now.
What doesn't work is a trigger with an M291 S2. After that, no more lines are executed.

P-C-R, 2020-08-31:
Yep. But the Initial Bug Report did Not contain a M291 ;) so i tested this first!

wolfgangmauer, 2020-08-31:
To be on the safe side, I tried it again
Nothing happens after the pause message.
Maybe DWC has problem with messages and acknowledge on SKR 1.3/SBC

wolfgangmauer, 2020-08-31:
After updating the binary file and executing LPC_DSF_update.
All works as expected!

closed 2020-08-31 by wolfgangmauer

P-C-R, 2020-08-31:
What you mean with updating the binary;) best

wolfgangmauer, 2020-08-31:
The firmware.bin, just to be sure ;-)
Working configuration:
config.g:
M950 J1 C"xstopmax"

start.g: <-- adding slicer startcode M98 P"0:/sys/start.g"
M581 P1 S1 T2
M582 T2

trigger2.g:
M581 S-1 T2
M25
M400
G4 P200              ; Wait 200 milliseconds
M83                  ; relative extruder moves
G1 E10 F500          ; Extrude 2mm to warm the heatbreak filament
G4 P500              ; Wait 500 milliseconds
G1 E-10 F500
G1 E-670 F3600       ; retract 670mm of filament
M400                 ; Wait for current moves to finish
M291 P"Insert Filament" R"Filament runout detected!" S2
M83                  ; relative extruder moves
G1 E670 F3600        ; extrude 670mm of filament
G1 E10 F500
M400                 ; Wait for current moves to finish
G4 P2000             ; Wait 2 seconds to clean nozzle
M581 P1 S1 T2        ; Reactivate trigger 2 
M582 T2              ; Check again there is filament 
M24

wolfgangmauer, 2020-08-31:
And even more comfort!

trigger2.g:
...
M291 P"Insert Filament" R"Filament runout detected!" S2
M83                  ; relative extruder moves
G1 E670 F3600        ; extrude 670mm of filament
G1 E10 F500
M400                 ; Wait for current moves to finish
**M98 P"/macros/purge_more"**
M581 P1 S1 T2        ; Reactivate trigger 2 
M582 T2              ; Check again there is filament 
...

0:/macros/purge_more
while true
	M291 P"Purge more filament?" R"Filament runout detected!" S3
	G1 E10 F500
	M400 ; Wait for current moves to finish

wolfgangmauer, 2020-08-31:
Is it possible to switch off the extruder temperature after an M25 (pause) and reset it to the original value on M24 (resume)?