gloomyandy/RepRapFirmware #37: [FR] Please provide PlatformIO(Windows/Linux) or makefile (Linux).

wolfgangmauer, 2020-08-31:
It would be very nice if there was a PlatformIO project or makefile.

wolfgangmauer, 2020-08-31:
Just begin with [gloomyandy/CoreLPC4RRF"](https://github.com/gloomyandy/CoreLPC4RRF) and everything works

gloomyandy, 2020-09-01:
Sorry I really do not like PlatformIO, I see no real benefit in using it and my experience of it with Marlin is that it is basically the source of many problems (with versions of components/tools changing without people realising plus numerous odd problems that seem to require deleting and reinstalling various parts of platformio). 

The build system for this project will be changing considerably as I plan to bring the LPC and STM32F4 ports together into a more unified version.

wolfgangmauer, 2020-09-01:
Sorry, it was my fault ;-)
I missed CoreL PC4 REF, there is .sh to create the Hole project.
Everything is fine.

gloomyandy, 2020-09-01:
No problem. This is all changing pretty quickly at the moment I hope to make it a little easier in the future but it all takes time!

closed 2020-09-01 by gloomyandy