gloomyandy/RepRapFirmware #41: [BUG] calibrationDeviation unknown

wolfgangmauer, 2020-09-05:
I found a example [bed.g](https://duet3d.dozuki.com/Wiki/GCode_Meta_Commands#Section_Using_conditional_GCode_commands_in_bed_g_to_calibrate_a_delta_printer) in the Wiki to calibrate the bed.
If i run a G32 command i get the following output

> G32
Leadscrew adjustments made: 0.120 0.119, points used 2, (mean, deviation) before (0.119, 0.000) after (0.000, 0.000)
Error: in file macro line 19 column 6: meta command: unknown value 'calibrationDeviation'

My bed.g
```; 0:/sys/bed.g
M561                    ; clear any bed transform
; If the printer hasn't been homed, home it
if !move.axes[0].homed || !move.axes[1].homed || !move.axes[2].homed
  G28
while true
  if iterations = 5
    abort "Too many auto calibration attempts"
  G30 P0 X20 Y117.5 Z-99999 ; probe near a leadscrew, half way along Y axis
  if result != 0
    continue
  G30 P1 X215 Y117.5 Z-99999 S2 ; probe near a leadscrew and calibrate 2 motors
  if result != 0
    continue
  if move.calibrationDeviation.deviation <= 0.02
    break
  echo "Repeating calibration because deviation is too high (" ^ move.calibrationDeviation.deviation ^ "mm)"
echo "Auto calibration successful, deviation", move.calibrationDeviation.deviation ^ "mm"
G29  ; probe the bed and enable compensation
```

jaysuk, 2020-09-05:
move.calibration.final.deviation

wolfgangmauer, 2020-09-05:
Many thanks, so the example is incorrect...

closed 2020-09-05 by wolfgangmauer