gloomyandy/RepRapFirmware #44: Max3155 temperature issue

overdrive2max, 2020-09-10:
I am running an SKR 1.3 board with sbc and a max3155 chip for the hotend. 
The temperature is very stable if I heat the hotend and manually move the printer axis. Turning on the heatbed does not impact on the readings. 
Once I start a print almost every second temperature reading is 2000c. Sometimes the print will fail due to disconnect being registered.
Any advice please?

board.txt
leds.diagnostic = NoPin; 

//Internal SDCard SPI Frequency.
lpc.internalSDCard.spiFrequencyHz  = 25000000;

heat.tempSensePins = {bedtemp}; //Max of 3 entries
//Soft SPI Temp sensor to use with MAX31865+PT100
softwareSPI.pins = {1.18,1.19,1.20}
heat.spiTempSensorCSPins = {1.21, NoPin }; //Max of 2 entries
heat.spiTempSensorChannel = 2
	 
//Analogue to Digital prefilter
adc.prefilter.enable = true
sbc.lpcTfrReadyPin = 1.31

//TMC Smart Drivers
stepper.numSmartDrivers = 4; //Adjust to the correct number of smart drivers
stepper.TmcDiagPins = { NoPin, NoPin, NoPin, NoPin, NoPin }

config.g
M308 S1 P"P1.21" Y"thermocouple-max31855"

overdrive2max, 2020-09-10:
I should also note that using a thermistor on the same cable moved to the analog input works without any issue. I have largely shielded the cables from stay voltages and to some extent capacitance that could creep in on anything sensor related.

gloomyandy, 2020-09-11:
Hi, Please post all of your config.g file, the output from M122 p200 and the output from M122. Also please post a photo of your board and how you have it set up. How long is the connection to the SBC?

Can you clarify what you mean by "Sometimes the print will fail due to disconnect being registered"? What exactly happens and what error message do you get? Next time you get that error, please post the output from M122 as soon after the error as you can. 

It sounds like something during the print is causing the SPI connection to the max3155 to read incorrect results. This could either be a software problem or some sort of hardware problem (noise or possibly a bad connection). Things to try....
1. Try simulating a print job and monitor the temperature during the simulation (this will create a lot of traffic over the SBC connection and we can check if that is causing the problem).
2. Please try manually turning on any software controlled fans and see if they trigger the problem.
3. Please check to see if you see bad readings during home all moves, or if you have a probe during G29 probing.
4. How long are the wires from the SKR board to the max3155?
5. Try creating a test gcode file that only has X/Y moves in it (no heaters and no extruding). Try running that file and see if you get the error.
6. If you do get the error in 5 above and have a spare thrmocouple try attaching the spare thermocouple and run the test again.
7. If you do not get the error in 5 try adding Z moves into the test file.

I'm pretty busy at the moment and do not have max3155 hardware to test with so this may take some time to work through. You might want to join the discord server (see the wiki) and ask over there is anyone else is using max3155 and if they are seeing problems.

gloomyandy, 2020-09-11:
One other question. Have you tried running the same print with a standard temperature sensor? It would be useful to know if you get any errors/disconnects in this case. It is possible that you have more than one problem here (the temperature reading and the disconnect).

overdrive2max, 2020-09-11:
config.g

; https://duet3d.dozuki.com/Wiki/Gcode

; General preferences
G90                                            ; send absolute coordinates...
M83                                            ; ...but relative extruder moves
M550 P"Zippy"                                     ; set printer name

M667 S1                                        ; select CoreXY mode

; Drives
M569 P0 S1 T0.1:0.1:0.02:0.02 D2                                  ; physical drive 0 goes forwards using default driver timings
M569 P1 S1 T0.1:0.1:0.02:0.02 D2                                      ; physical drive 1 goes forwards using default driver timings
M569 P2 S0 T0.1:0.1:0.02:0.02 D2                                ; physical drive 2 goes forwards using default driver timings
M569 P3 S0 T0.1:0.1:0.02:0.02 D2                                    ; physical drive 3 goes forwards using default driver timings

M584 X0 Y1 Z2 E3                              ; set drive mapping

M350 X16 Y16 Z16 E16 I1                        ; configure microstepping with interpolation
M92 X80 Y80 Z160.45 E430         ; set steps per mm
M566 X900.00 Y900.00 Z60.00 E2000.00 P1           ; set maximum instantaneous speed changes (mm/min)
M203 X30000.00 Y30000.00 Z1000.00 E3000.00        ; set maximum speeds (mm/min)
M201 X5000.00 Y5000.00 Z120.00 E1000.00           ; set accelerations (mm/s^2)
M204 P1000.00 T5000.00
M906 X1000 Y1000 Z1200 E900 I70                  ; set motor currents (mA) and motor idle factor in per cent
;M84 S20                                        ; Set idle timeout
M593 F40.5										;M593: Configure Dynamic Acceleration Adjustment

; Axis Limits
M208 X0 Y0 Z0 S1                           ; Set axis max travel
M208 X298 Y300 Z300 S0                         ; set axis maxima

; Endstops
M574 X1 S1 P"xstop"                            ; configure active-high endstop for low end on X via pin xstop
M574 Y1 S1 P"ystop"                            ; configure active-high endstop for low end on Y via pin ystop
M574 Z1 S2                                     ; configure Z-probe endstop for low end on Z

; Z-Probe
M558 P5 C"zstop" H2 F480 T9000 R1 A10 S0.03          ; set Z probe type to switch and the dive height + speeds
G31 P300 X-31.7 Y0 Z10.2                           ; set Z probe trigger value, offset and trigger height
M557 X40:250 Y40:250 P5                      ; define mesh grid

; Heaters
M308 S0 P"bedtemp" Y"thermistor" T100000 B3950 ; configure sensor 0 as thermistor on pin bedtemp
M950 H0 C"bed" T0                              ; create bed heater output on bed and map it to sensor 0
M307 H0 B0 S1.00                               ; disable bang-bang mode for the bed heater and set PWM limit
M140 H0                                        ; map heated bed to heater 0
M143 H0 S130                                   ; set temperature limit for heater 0 to 130C
;M308 S1 P"th1" Y"thermistor" T98014 B4388 ; configure sensor 1 as thermistor on pin e0temp
M308 S1 P"P1.21" Y"thermocouple-max31855"  ; create sensor number 1 as a thermocouple, defaulting to type K in the first position on the Duet 2 ;daughter board connector
M950 H1 C"he0" T1                           ; create nozzle heater output on e0heat and map it to sensor 1
M307 H1 B0 S1.00                               ; disable bang-bang mode for heater  and set PWM limit

; Fans
M950 F0 C"fan0" Q500                           ; create fan 0 on pin fan0 and set its frequency
M106 P0 S0 H-1                                 ; set fan 0 value. Thermostatic control is turned off
M950 F1 C"e1heat" Q500                            ; create fan 1 on pin he1 and set its frequency
M106 P1 H1 T50                              ; set fan 1 value. Thermostatic control is turned on at 50 deg C

; Tools
M563 P0 D0 H1 F0                               ; define tool 0
G10 P0 X0 Y0 Z0                                ; set tool 0 axis offsets
G10 P0 R0 S0                                   ; set initial tool 0 active and standby temperatures to 0C

; Custom settings are not defined
T0  ;select tool
M501 

I made a daughter board to connect the Pi and Max31855 to the ext connectors of the SKR. I added buck converters to supply 12v and 5v from the 24v supply and a linear regulator to supply 3.3v stable.
[... 71 lines cut ...]
=== Linux interface ===
State: 0, failed transfers: 0
Last transfer: 11ms ago
RX/TX seq numbers: 49232/49234
SPI underruns 0, overruns 0
Number of disconnects: 0
Buffer RX/TX: 0/0-0
=== Duet Control Server ===
Duet Control Server v3.1.1
Code buffer space: 2048
Configured SPI speed: 8000000 Hz
Full transfers per second: 32.79

My cable run to the hotend runs past my X motor and the E motor (Not ideal I know). When X, Y and Z is moving there is no issue.

The disconnect becomes more frequent when the E stepper motor is engaged (moving or idle). 
When I print, most of the time it does not fail. The temperature spike (disconnect condition) lasts for 1 cycle on the graph by the looks of it. Sometimes it is longer and the print will fail due to heater failure condition. 

Using a 100k thermistor has no issue at all, everything works.

What bugs me is that it should not be interference causing my issue as the K-type readings is voltage based. Thus, emi would cause erratic temperatures, not a disconnected probe reading of 2000C.

It feels like the software SPI is not functioning or reporting fast enough.

Kind regards

sdavi, 2020-09-11:
Yes, most likely it's cause your using Software SPI for the max31865 connection. It may not like the variations in the software generated signals caused by the extra interrupts that are used for stepper movement when printing. Most users use the max31865 connected SSP0 on the LPC port (myself included), but since you also have the SBC connected, the only other choice is SSP1 (if your board has headers for those pins).

overdrive2max, 2020-09-11:
That was my suspicion. Can the ss SPI rate not be lowered to achieve stability?

overdrive2max, 2020-09-11:
I have to apologise for not doing proper research on this issue. It seems it was brought up on the RRF forums and a fix was proposed. I can confirm it has solved my issue on initial testing. 
By adding a bypass cap of 10nf per thermocouple terminal to ground solved my issue.

https://forum.duet3d.com/topic/1270/thermocouple-issues/11

gloomyandy, 2020-09-11:
@overdrive2max Thanks for the update. Please let us know if it still seems to be working after further testing.

closed 2020-09-27 by gloomyandy