gloomyandy/RepRapFirmware #45: LPC - TMC drivers

Aeronut, 2020-09-13:
I appreciate that this is in its infancy, but will other TMC options become available, 5160 for example?
Thanks in advance.

P-C-R, 2020-09-14:
Dear @Aeronut the TMC5160/5161 are SPI controlled drivers. I think its on the "roadmap" but not the most important thing to implement!

wolfgangmauer, 2020-09-15:
I'm waiting for TMC2130 support.

P-C-R, 2020-09-15:
But what could the 2130 do what the 2209 cant?;)

wolfgangmauer, 2020-09-15:
Possibly nothing, but that's the one i have ;-)

gloomyandy, 2020-09-16:
As P-C-R commented 5160 support is on my list of things to do, but I do not have any 5160s (or really any need for them), so it may be a while before I do anything in that area.

Nathan116, 2020-10-02:
> […](#)

Well, to give a simple answer, I don't have any 2209s but I do have a set of 2130s... With that being said, it looks like I'm going to be ordering some 2209s

sl1pkn07, 2020-11-16:
Hi. any update to support 2130SPI? (with sensorless) i have actually run Marlin with this drivers in my Rearm+Raps1.4. but i want to switch to RRF (i also have a paneldue and the wifi dongle for it)

greetings

gloomyandy, 2020-11-16:
Very unlikely I will add 2130 support. I have 5160 support on my "to do" list as they seem to offer something that 2209s do not and there is already support for them in the RRF codebase (there isn't for 2130s).

gloomyandy, 2021-03-26:
TMC5160 support is now available (from v3.3-beta1 onwards).

closed 2021-03-26 by gloomyandy