gloomyandy/RepRapFirmware #46: TODO list

gloomyandy, 2020-09-16:
So this is my current list of the things I'm planning on working on (in a very rough priority order)...
1. Create single code base for STM/LPC versions and associated build system.
2. Track DC42 3.2 beta and beyond.
3. Add support for various expansion boards (GTR M5 and BTT expansion board).
4. Add option to control initial state of PS_ON.
5. Enable support for NeoPixels.
6. Add 5160 support.
7. SDIO support for SD cards.
8. Improved pin configuration (avoid need to list all pins in the various board.h files).
Items 1 and 2 are my main focus at the moment. 3.2-beta1 is currently in test for most configurations.

I tend to work on things that interest me and that are easy for me to set up (so if I don't have the hardware and it is not interesting/cheap I'm unlikely to work on it). I am also in the middle of setting up a new printer and have plenty of of other things to do!

arrowcircle, 2020-09-22:
Donations link?

P-C-R, 2020-09-25:
https://ko-fi.com/rrfonlpcstmproject

mikerocklewitz, 2020-10-08:
Im experiencing an SD issue on brand new 1.4 turbo.  I get everything going and everything, for some reason, boots up.  but nothing I change in rrf config's on the sd card sticks. Duet webcontrol says fails to save/upload. all settings changed are gone on power cycle.  Sandisk microsd.  Just tried building marlin and it works....about to try this again with a different SD :)

TpyoKnig, 2020-10-08:
I think this should go into its own issue, but FWIW I had this issue when I updated the WiFi firmware when using the BTT RRF WiFi add on, downgrading the firmware back resolved the issue

> Im experiencing an SD issue on brand new 1.4 turbo.  I get everything going and everything, for some reason, boots up.  but nothing I change in rrf config's on the sd card sticks. Duet webcontrol [...]

Mushmuch, 2020-10-18:
I'm looking forward to the GTR M5 support to convert a Tenlog D3S IDEX with 7 steppers and 5 fans. Let me know if you need a M5 board.

P-C-R, 2020-10-18:
@Mushmuch 
M5 ist supported in the newest Beta

https://github.com/gloomyandy/RepRapFirmware/blob/v3.02-dev-unified/WHATS_NEW_UNIFIED.md

Minims, 2020-10-22:
is sensorsless with tmc2209 on SKR Pro/GTR will be available in the next beta ? Thx

jaysuk, 2020-10-22:
@Minims it works now

Minims, 2020-10-22:
Ok so I might do something wrong. Where can I get support for this ?

AndyRap, 2021-04-04:
Is there an Eta for adding SKR2 support? Thanks for all your hard work 👍

P-C-R, 2021-04-04:
Is the skr 2 Out?

Best

AndyRap, 2021-04-04:
> […](#)

Yes sir. Ordered two from the BigTree-Tech.com website this afternoon 👍

VinnyCordeiro, 2021-04-04:
There's already hardware info at Bigrtreetech's repository: https://github.com/bigtreetech/SKR-2

EsserPrototyping, 2021-04-05:
> Yes sir. Ordered two from the BigTree-Tech.com website this afternoon

Thanks AndyRap.. just grabbed one from the store for the limited price, which is not aviable anymore on aliexpress:)

sq4nyy, 2021-04-25:
BTT now doesn't provide packs to flash and links to your github. If you cooperate with them, could you please provide ready to flash packs for SKR E3 RRF?

jaysuk, 2021-04-25:
We already do. The SKR-RRF-E3 is part of the config tool and has been
supported since release 3.3b1_5

sq4nyy, 2021-04-25:
I'm still doing something wrong or this is not a ready solution for noobs being in need just to make their skr e3 rrf work, There's no stock lcd support. BTT screwed Marlin which does not work properly and they do not even post ready single zip pack to flash RRF for those who do not want to graduate from new firmware and I am one of them.

jaysuk, 2021-04-25:
It works fine on my SKR-RRF-E3 + IDEX.
We have instructions here
https://teamgloomy.github.io/skr_rrf_e3_general.html
There are videos on YouTube
And the 12864 displays do work, as well as TFTs and other displays. The
only ones that don't work are the Ender3v2 as it uses a custom firmware.
We also offer a github repository for the fly-E3 board which would work
with the board.txt file for an SKR-RRF-E3 as the naming is the same.
You can also select the ender 3 config on the config tool and swap the
board to use the SKR-RRF-E3

gloomyandy, 2021-04-25:
Please continue this discussion ether over on discord or in a new issue, please respect my request to not post comments on this issue.