gloomyandy/RepRapFirmware #48: No web interface after clean install on SBC

overdrive2max, 2020-09-22:
Hi
I am using an SKR 1.3 via SBC connection to a RPI 3A+
It was working fine and I was doing an update.
I ran the update as per the wiki:
 wget https://raw.githubusercontent.com/P-C-R/LPC-DSF-Install_Script/master/LPC_DSF_update.sh
sudo chmod 755 LPC_DSF_update.sh
./LPC_DSF_update.sh

after the update to 3.2.0-beta+1 The browser just displays connection refused.
I then re-flashed the sd card with duetpi and tried every one of the stable/unstable wiki instructions with no luck.
Am I missing something here?

jaysuk, 2020-09-22:
Please follow the correct instructions when updating to beta 3.2 on an LPC
https://github.com/gloomyandy/RepRapFirmware/wiki/SBC-Update-LPC-Unstable

overdrive2max, 2020-09-22:
Hi
Same result. 
Either using duetpi lite or backup image of my printer's RPi on 3.1.1

I also always get an error with "sudo rm /etc/apt/sources.list.d/duet3d.list"
 
pi@Zippy:~ $ wget -q https://pkg.duet3d.com/duet3d.gpg
pi@Zippy:~ $ wget -q https://pkg.duet3d.com/duet3d-unstable.list
pi@Zippy:~ $ sudo mv duet3d.gpg /etc/apt/trusted.gpg.d/
pi@Zippy:~ $ sudo mv duet3d-unstable.list /etc/apt/sources.list.d/duet3d-unstable.list
pi@Zippy:~ $ sudo chown root:root /etc/apt/trusted.gpg.d/duet3d.gpg
pi@Zippy:~ $ sudo chown root:root /etc/apt/sources.list.d/duet3d-unstable.list
pi@Zippy:~ $ sudo rm /etc/apt/sources.list.d/duet3d.list
rm: cannot remove '/etc/apt/sources.list.d/duet3d.list': No such file or directory

jaysuk, 2020-09-22:
thats fine because you've probably changed over to the unstable channel previously. finish following the instructions

overdrive2max, 2020-09-22:
Still no web interface sadly
I see the web server fails:
sudo systemctl status  duetwebserver
● duetwebserver.service - Duet Web Server
   Loaded: loaded (/lib/systemd/system/duetwebserver.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Tue 2020-09-22 10:11:56 SAST; 18min ago
  Process: 4091 ExecStart=/opt/dsf/bin/DuetWebServer (code=killed, signal=ABRT)
 Main PID: 4091 (code=killed, signal=ABRT)

Sep 22 10:11:56 Zippy DuetWebServer[4091]:    at Microsoft.Extensions.DependencyInjection.ServiceLookup.DynamicServiceProviderEngine.<>c__Di
Sep 22 10:11:56 Zippy DuetWebServer[4091]:    at Microsoft.Extensions.DependencyInjection.ServiceLookup.ServiceProviderEngine.GetService(Typ
Sep 22 10:11:56 Zippy DuetWebServer[4091]:    at Microsoft.Extensions.DependencyInjection.ServiceLookup.ServiceProviderEngineScope.GetServic
Sep 22 10:11:56 Zippy DuetWebServer[4091]:    at Microsoft.Extensions.DependencyInjection.ServiceProviderServiceExtensions.GetService[T](ISe
Sep 22 10:11:56 Zippy DuetWebServer[4091]:    at Microsoft.Extensions.Hosting.Internal.Host.StartAsync(CancellationToken cancellationToken)
Sep 22 10:11:56 Zippy DuetWebServer[4091]:    at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, Cancell
Sep 22 10:11:56 Zippy DuetWebServer[4091]:    at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.Run(IHost host)
Sep 22 10:11:56 Zippy DuetWebServer[4091]:    at DuetWebServer.Program.Main(String[] args) in /home/christian/Duet3D/DuetSoftwareFramework/s
Sep 22 10:11:56 Zippy systemd[1]: duetwebserver.service: Main process exited, code=killed, status=6/ABRT
Sep 22 10:11:56 Zippy systemd[1]: duetwebserver.service: Failed with result 'signal'.

jaysuk, 2020-09-22:
something within your upgrade process has failed and its best to start from a fresh duetpi image.
the contents of the SD card can be round at /opt/dsf/sd

overdrive2max, 2020-09-22:
"I then re-flashed the sd card with duetpi and tried every one of the stable/unstable wiki instructions with no luck."
I have tried all combinations of the instructions from the wiki applied to my backup cloned images and vanilla duetpi lite image.
Same result on all

jaysuk, 2020-09-22:
I have just followed the instructions detailed here https://github.com/gloomyandy/RepRapFirmware/wiki/SBC-Update-LPC-Unstable
with a fresh duetpi-lite image from here https://pkg.duet3d.com/DuetPi-lite.zip
I followed the instructions, switch from the stable branch to the unstable branch and completed the upgrade without issue.
I was also able to modify the config json (after changing sudo systemctl Stop duetcontrolserver to sudo systemctl stop duetcontrolserver).
This resulted in a working image with functioning web interface.

overdrive2max, 2020-09-22:
ok, I Re-downloaded duetpi lite and used followed the procedure and the web interface is working. This is weird as etcher would give errors if the archive is corrupt.

Now I have a new problem:  v3.2-beta1_2 - firmware-lpc-sbc-3.2-beta1_2.bin does not work. I get the transfer pin not ready error. going back to 3.1.1-14 works.

jaysuk, 2020-09-22:
did you make the two json changes?
I also use that image and it works fine for me.
we also have a number of other users that have upgraded and so far, you're the only one reporting issues

overdrive2max, 2020-09-22:
yes, as changing the skr firmware back to 3.1.1 works, the RPI communicates with the SKR no issue. only when updating the firmware I get the error.

jaysuk, 2020-09-22:
then you must be doing something wrong

overdrive2max, 2020-09-22:
how? 
download firmware-sbc.bin  3.1.1-14 from:
https://github.com/gloomyandy/RepRapFirmware/releases
rename to firmware.bin
copy firmware.bin to sd card
insert sd card
restart skr
verify firmware updated

Works!

download firmware-lpc-sbc-3.2-beta1_2.bin  3.2-beta1_2 from:
https://github.com/gloomyandy/RepRapFirmware/releases
rename to firmware.bin
copy firmware.bin to sd card
insert sd card
restart skr
verify firmware updated

[image]

jaysuk, 2020-09-22:
its fine for everyone else including myself running SKR boards with SBC.
If you have further issues join discord

closed 2020-09-22 by jaysuk

TpyoKnig, 2020-09-27:
FWIW I ran into this same issue. I troubleshot this down to be the linked shell script for the stable build actually installs the unstable build of DWC/DCS but then installs the stable patches resulting in a nonfunctional UI with an error relating to the assembly references mismatch with the DuetAPI.dll. 

The fix is to use the script but omit the portion where it downloads the unstable DWC/DCS and it works as expected.

P-C-R, 2020-09-27:
@overdrive2max @TpyoKnig 

Thank you for pointing it out! Fixed it