gloomyandy/RepRapFirmware #49: Touchscreen support

cotestatnt, 2020-09-22:
Hi! thank you for your great job, I really love this firmware.
I'm trying to get a BTT touch screen display to work on a SKR 1.4.
Unfortunately for now without success.

It seems that the serial port don't works as it should (clearly after enabling it with M575 P1 B115200 S1 in config.g).
I also tried with a classic USB / TTL converter but no reply from board.

What am I doing wrong?

jaysuk, 2020-09-22:
have you read this page?
https://github.com/gloomyandy/RepRapFirmware/wiki/SKR-PanelDue

There is also a slightly newer TFT specific firmware rather than using bcmobs.
it can be found at https://github.com/JimmymaDdness/Mks-TouchScreenFirmware-Duet

cotestatnt, 2020-09-22:
Damn!
Yes, but maybe i've missed the last step in board.txt

Thank you @jaysuk, i will try as soon as i can!

closed 2020-09-29 by jaysuk