gloomyandy/RepRapFirmware #50: BABY STEPS not working

MontyFlange, 2020-09-25:
New install , no matter how many times I press the move closer button nothing happens. Took it down 3.0 mm and it never moved 

Board: biquskr_1.3 (LPCWiFi)
Firmware: RepRapFirmware for LPC176x based Boards 3.1.1-14 (2020-17-7b14)
Duet WiFi Server Version: 1.23-01L-D

P-C-R, 2020-09-25:
Please upload you config.g

MontyFlange, 2020-09-25:
``; Configuration file for SKR v1.3 (firmware version 3)
; executed by the firmware on start-up
;
; generated by RepRapFirmware Configuration Tool v3.1.3-LPC-1 on Thu Aug 20 2020 00:30:00 GMT+0100 (British Summer Time)

; General preferences
G90                                            ; send absolute coordinates...
M83                                            ; ...but relative extruder moves
M550 P"HAL 2020"                               ; set printer name

; Network
M552 S1                                        ; enable network 
M586 P0 S1                                     ; enable HTTP
M586 P1 S0                                     ; disable FTP
M586 P2 S0                                     ; disable Telnet
M575 P1 S1 B57600                              ; enable TFT

; Drives
M569 P0 S0 D3 V26                                   ; X motor physical drive 0 goes forwards using default driver timings
M569 P1 S0 D3 V26                                   ; Y Motor physical drive 1 goes forwards using default driver timings
M569 P2 S1 D3 V26                                   ; Z1 Motor physical drive 2 goes forwards using default driver timings
M569 P3 S0 D3 V26                                   ; Extruder Motor physical drive 3 goes forwards using default driver timings
M569 P4 S1 D3 V26                                   ; Z2 
M584 X0 Y1 Z2:4 E3                                  ; set drive mapping

M350 X16 Y16 Z16 E16 I1                        ; configure microstepping with interpolation
M92 X80.00 Y80.00 Z400.00 E99.00               ; set steps per mm
M566 X900.00 Y900.00 Z12.00 E120.00            ; set maximum instantaneous speed changes (mm/min)
M203 X6000.00 Y6000.00 Z180.00 E1200.00        ; set maximum speeds (mm/min)
M201 X500.00 Y500.00 Z20.00 E250.00            ; set accelerations (mm/s^2)
M906 X800 Y800 Z800 E800 I30                   ; set motor currents (mA) and motor idle factor in per cent
M84 S30                                        ; Set idle timeout

; Axis Limits
M208 X0 Y0 Z0 S1                               ; set axis minima
M208 X310 Y300 Z395 S0                         ; set axis maxima

; Endstops
M574 X2 S1 P"Xstop"                            ; configure active-high endstop for low end on X via pin ystop
M574 Y1 S1 P"ystop"                            ; configure active-high endstop for low end on Y via pin ystop
M574 Z1 S1 P"zstop"                            ; configure active-high endstop for low end on Z via pin zstop

; Heaters
M308 S0 P"bedtemp" Y"thermistor" T100000 B4138 ; configure sensor 0 as thermistor on pin bedtemp
M950 H0 C"bed" T0                              ; create bed heater output on bed and map it to sensor 0
M140 H0                                        ; map heated bed to heater 0
M143 H0 S120                                   ; set temperature limit for heater 0 to 120C
M307 H0 A93.8 C545.0 D1.6 S1.00 V12.3 B0
M307 H1 A349.3 C132.0 D6.0 S1.00 V0.0 B0
M308 S1 P"e0temp" Y"thermistor" T100000 B4138  ; configure sensor 1 as thermistor on pin e0temp
M950 H1 C"e0heat" T1                           ; create nozzle heater output on e0heat and map it to sensor 1

; Fans
M950 F0 C"e1heat" Q500                         ; create fan 0 on pin fan0 and set its frequency
M106 P0 S0 H-1                                 ; set fan 0 value. Thermostatic control is turned off

; Tools
M563 P0 D0 H1 F0                               ; define tool 0
G10 P0 X0 Y0 Z0                                ; set tool 0 axis offsets
G10 P0 R0 S0                                   ; set initial tool 0 active and standby temperatures to 0C
T0                                             ; select first tool
M561                                           ; Cleard Bed Levelling

jaysuk, 2020-09-26:
Your axis minima for Z is 0. (M208)
The firmware won't let you go any further than that.
Lower that value.

MontyFlange, 2020-09-26:
Great thanks, Will report back

closed 2020-09-29 by jaysuk