gloomyandy/RepRapFirmware #51: SKR 1.3 reset when starting a job

overdrive2max, 2020-10-09:
Hi
Randomly, when I have completed a job and start a new one the SKR reboots. Sometimes this happens twice in a row then it will print just fine. This happens on 3 of my machines with the SKR 1.3 in SBC configuration with a Pi 3 A+
All have stable meanwell power supplies and Pi and aux power is connected via bespoke pcb.
Any advice would be appreciated.

My config.g of one of the smaller machines

`G90                                            ; send absolute coordinates...
M83                                            ; ...but relative extruder moves
M550 P"Helios"                                     ; set printer name

M667 S1                                        ; select CoreXY mode

; Drives
M569 P0 S1 T0.1:0.1:0.02:0.02 D2                                  ; physical drive 0 goes forwards using default driver timings
M569 P1 S1 T0.1:0.1:0.02:0.02 D2                                      ; physical drive 1 goes forwards using default driver timings
M569 P2 S0 T0.1:0.1:0.02:0.02 D2                                ; physical drive 2 goes forwards using default driver timings
M569 P3 S0 T0.1:0.1:0.02:0.02 D2                                    ; physical drive 3 goes forwards using default driver timings

M584 X0 Y1 Z2 E3                              ; set drive mapping

M350 X16 Y16 Z16 E16 I1                        ; configure microstepping with interpolation
M92 X80 Y80 Z200 E430         ; set steps per mm
M566 X900.00 Y900.00 Z60.00 E2000.00 P1           ; set maximum instantaneous speed changes (mm/min)
M203 X30000.00 Y30000.00 Z2000.00 E3000.00        ; set maximum speeds (mm/min)
M201 X5000.00 Y5000.00 Z120.00 E1000.00           ; set accelerations (mm/s^2)
M204 P1000.00 T5000.00
M906 X1000 Y1000 Z1200 E900 I70                  ; set motor currents (mA) and motor idle factor in per cent
;M84 S20                                        ; Set idle timeout
M593 F40.5										;M593: Configure Dynamic Acceleration Adjustment

; Axis Limits
M208 X-17 Y-29 Z0 S1                           ; Set axis max travel
M208 X330 Y330 Z300 S0                         ; set axis maxima

; Endstops
M574 X1 S1 P"xstop"                            ; configure active-high endstop for low end on X via pin xstop
M574 Y1 S1 P"ystop"                            ; configure active-high endstop for low end on Y via pin ystop
M574 Z1 S2                                     ; configure Z-probe endstop for low end on Z

; Z-Probe
M558 P9 H2 F680 T8000 X0 Y0 Z1 C"^zstop" A10 S0.03    ; disable Z probe but set dive height, probe speed and travel speed
M950 S0 C"servo0"                              ; Setup servo 0 as servo port on SKR

;M558 P5 C"zstop" H2 F480 T9000 R1 A10 S0.03          ; set Z probe type to switch and the dive height + speeds
G31 P300 X-23.5 Y0 Z1.25                          ; set Z probe trigger value, offset and trigger height
M557 X40:290 Y40:290 P5                      ; define mesh grid

; Heaters
M308 S0 P"bedtemp" Y"thermistor" T100000 B3950 ; configure sensor 0 as thermistor on pin bedtemp
M950 H0 C"bed" T0                              ; create bed heater output on bed and map it to sensor 0
M307 H0 B0 S1.00                               ; disable bang-bang mode for the bed heater and set PWM limit
M140 H0                                        ; map heated bed to heater 0
M143 H0 S130                                   ; set temperature limit for heater 0 to 130C
M308 S1 P"th1" Y"thermistor" T95002.69 B4322.60 ; configure sensor 1 as thermistor on pin e0temp
;M308 S1 P"P1.21" Y"thermocouple-max31855"  
M950 H1 C"he0" T1                           ; create nozzle heater output on e0heat and map it to sensor 1
M307 H1 B0 S1.00                               ; disable bang-bang mode for heater  and set PWM limit

; Fans
M950 F0 C"fan0" Q500                           ; create fan 0 on pin fan0 and set its frequency
M106 P0 S0 H-1                                 ; set fan 0 value. Thermostatic control is turned off
M950 F1 C"e1heat" Q500                            ; create fan 1 on pin he1 and set its frequency
M106 P1 H1 T50                              ; set fan 1 value. Thermostatic control is turned on at 50 deg C

; Tools
M563 P0 D0 H1 F0                               ; define tool 0
G10 P0 X0 Y0 Z0                                ; set tool 0 axis offsets
G10 P0 R0 S0                                   ; set initial tool 0 active and standby temperatures to 0C

; Custom settings are not defined
T0  ;select tool
M501 `

boards.txt
`lpc.board = biquskr_1.3; 

//LED blinks to indicate Platform is spinning or other diagnostic 
//Comment out or set to NoPin if not wanted.
leds.diagnostic = NoPin; 

//Internal SDCard SPI Frequency.
lpc.internalSDCard.spiFrequencyHz  = 25000000;

//LCD Pins (only ST9720 SPI currently supported)
//setup the Software SPI Pins for LCD

heat.tempSensePins = {bedtemp, e0temp}; //Max of 3 entries
//Soft SPI Temp sensor to use with MAX31865+PT100
//softwareSPI.pins = {1.18,1.19,1.20}
//heat.spiTempSensorCSPins = {1.21, NoPin }; //Max of 2 entries
//heat.spiTempSensorChannel = 2
	 
//Analogue to Digital prefilter
adc.prefilter.enable = true
sbc.lpcTfrReadyPin = 1.31
//serial.aux.rxTxPins = { NoPin, NoPin }

//TMC Smart Drivers
stepper.enablePins = { 2.1, 2.8, 0.21, 2.12, 0.10 }
stepper.stepPins = { 2.2, 0.19, 0.22, 2.13, 0.1 }
stepper.directionPins = { 2.6, 0.20, 2.11, 0.11, 0.0 }
stepper.digipotFactor = 0.00
stepper.TmcUartPins = { 1.17, 1.15, 1.10, 1.8, 1.1 }
stepper.numSmartDrivers = 4; //Adjust to the correct number of smart drivers
stepper.TmcDiagPins = { NoPin, NoPin, NoPin, NoPin, NoPin }
`

jaysuk, 2020-10-10:
What version are you using?

overdrive2max, 2020-10-10:
Duet Web Control 3.1.1
Board: biquskr_1.3 (LPCSBC)
DSF Version: 3.1.1
Firmware: RepRapFirmware for LPC176x based Boards 3.1.1-14 (2020-17-7b14)

gloomyandy, 2020-10-10:
Please use the discord server for support.

wolfgangmauer, 2020-10-30:
@overdrive2max I had exactly the same problem, with me it was a short circuit from the HE0 :-(

overdrive2max, 2020-10-30:
> @overdrive2max I had exactly the same problem, with me it was a short circuit from the HE0 :-(

Thanks, but no electrical issue here. Have been running Klipper since I posted my issue without any issue. Tried RRF 2 days ago and reset every 2nd or 3rd print.

CHaZNiK8, 2020-11-07:
I had the same problem with SKR 1.4 (SBC RPI 3B+). I solved this problem with increase power supply voltage form 24,1V to 24,4V. Power supply 24V shared withSKR1.4 and  RPI via 5V BEC.
Raspberry Pi is very sensitive to input voltage.

closed 2020-12-19 by gloomyandy