gloomyandy/RepRapFirmware #57: G1 + H1 moves only one Z motor

ScotchTape, 2020-11-22:
Hi,
I've set up my Anycubic i3 with Dual Z and Dual endstops, configured by the book
config.g sections:
`; Drives
M569 P0 S1 D3 V40                                             ; X physical drive 0 goes forwards using default driver timings
M569 P1 S0 D3 V40                                            ; Y physical drive 1 goes backwards using default driver timings
M569 P2 S0                                             ; Z1 physical drive 2 goes backwards using default driver timings
M569 P3 S1                                             ; E physical drive 3 goes forwards using default driver timings
M569 P4 S0                                             ; Z2 physical drive 4 goes backwards using default driver timings
M584 X0 Y1 Z2:4 E3                                       ; set drive mapping

M574 Z1 S1 P"!zstop+!pwrdet" 							   ; configure active-low endstops for low end on Z via pins zstop and e1stop`

However what I try, while homing Z only one motor is driven while the other one is energized and locked.
In essence I switch to relative position in homez.g and
```G1 H1 Z-210 F1800 ; move Z down until the endstop is triggered

If I just move the axis, everything is nice and dandy.
I don't understand it. Endstops are reported fine, the are assigned to the correct motors.

Bords is a SKR 1.4 running 3.1.1-14

P-C-R, 2020-11-23:
Can you Join the discord?

Have you disabled(Cut) the diag Pin on the z drivers?

ScotchTape, 2020-11-23:
Yes, I can join your discord if you tell me which one or send me an invite.
I don‘t need to cut the pins, since Z and E1 are 2208. Nevertheless homing was fine with Marlin and Klipper.

jaysuk, 2020-11-23:
https://discord.gg/uS97Qs7
Thanks,

Jay

ScotchTape, 2020-11-24:
As requested, here are the files (config.g, homing)

[configs.zip](https://github.com/gloomyandy/RepRapFirmware/files/5587113/configs.zip)

ScotchTape, 2020-11-26:
Fixed by moving from Wifi Setup to SBC

closed 2020-11-26 by ScotchTape