gloomyandy/RepRapFirmware #60: SKR PRO Print stop for no reason and marked as finished 100%

sankafola, 2020-12-15:
i have SKR PRO with SBCC PI 3B+ 

issues 

1-SKR PRO Print stop for no reason and marked as finished 100% its like the PI crash because i cant change anything in the DWC , i cant load any prints or reprint and i have to reboot both the SKR Pro and the PI 
2-when i do the auto calibration g32 when its done it says response too long but it give value as if it passes
3-G29 i get a red error saying errors then it says successful my M558 is M558 P8 H2 F1000 T12000 R0.2 S0.03 A5 C"servo0"

my system is

Board: biquskrpro_1.1 (STMSBC)
DSF Version: 3.2.0-beta4
Firmware: RepRapFirmware for STM32F4 based Boards 3.2-beta4.1_1 (2020-12-03)

printer is Anyquibic Predator Delta 

my m122 

m122
=== Diagnostics ===
RepRapFirmware for STM32F4 based Boards version 3.2-beta4.1_1 running on STM32F4 (SBC mode)
Used output buffers: 1 of 40 (10 max)
=== RTOS ===
Static ram: 26240
Dynamic ram: 77236 of which 76 recycled
Never used RAM 26492, free system stack 135 words
Tasks: Linux(blocked,115) HEAT(blocked,298) MAIN(running,653) IDLE(ready,20)
Owned mutexes: HTTP(MAIN)
=== Platform ===
Last reset 00:01:07 ago, cause: [power on/off]
Last software reset time unknown, reason: User, none spinning, available RAM 28024, slot 2
Software reset code 0x0012 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x00400000 BFAR 0xe000ed38 SP 0xffffffff Task Linu
Error status: 0x00
MCU temperature: min 36.0, current 37.8, max 38.1
Driver 0: position 124255, standstill, SG min/max 0/6, error r/w 0/0, ifcnt 14, cnt r/w 5814/14, timeout 0, failedOp 0xff
Driver 1: position 197892, standstill, SG min/max 0/8, error r/w 0/0, ifcnt 13, cnt r/w 5815/13, timeout 0, failedOp 0xff
Driver 2: position 124201, standstill, SG min/max 0/8, error r/w 0/0, ifcnt 13, cnt r/w 5814/13, timeout 0, failedOp 0xff
Driver 3: position 0, standstill, SG min/max 0/0, error r/w 0/0, ifcnt 11, cnt r/w 5816/11, timeout 0, failedOp 0xff
Driver 4: position 0
Driver 5: position 0
Driver 6: position 0
Driver 7: position 0
Driver 8: position 0
Driver 9: position 0
Driver 10: position 0
Date/time: 2020-12-15 02:13:22
Slowest loop: 95.75ms; fastest: 0.05ms
=== Move ===
FreeDm 100 (min 97), maxWait 29600ms, bed compensation in use: none, comp offset 0.000
=== DDARing ===
Scheduled moves 4, completed moves 4, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== Heat ===
Bed heaters = 0, chamberHeaters = -1
Heater 1 is on, I-accum = 0.0
=== GCodes ===
Segments left: 0
Movement lock held by null
HTTP* is doing "M122" in state(s) 0
Telnet is idle in state(s) 0
File is idle in state(s) 0
USB is idle in state(s) 0
Aux is idle in state(s) 0
Trigger* is idle in state(s) 0
Queue is idle in state(s) 0
LCD is idle in state(s) 0
SBC is idle in state(s) 0
Daemon is idle in state(s) 0
Aux2 is idle in state(s) 0
Autopause is idle in state(s) 0
Code queue is empty.
=== Filament sensors ===
Extruder 0 sensor: ok
=== SBC interface ===
State: 4, failed transfers: 0
Last transfer: 3ms ago
RX/TX seq numbers: 2149/2149
SPI underruns 0, overruns 0
Number of disconnects: 0, IAP RAM available 0x07178
Buffer RX/TX: 0/0-0
=== Duet Control Server ===
Duet Control Server v3.2.0-beta4
Code buffer space: 4096
Configured SPI speed: 8000000 Hz
Full transfers per second: 35.61

P-C-R, 2020-12-15:
Please post the config.g, bed.g

sankafola, 2020-12-15:
hi here is the config.g and bed.g 

; Configuration file for SKR v1.4 (firmware version 3)
; executed by the firmware on start-up
;
; generated by RepRapFirmware Configuration Tool v3.1.3-LPC-1 on Sat Aug 22 2020 07:45:45 GMT-0400 (Eastern Daylight Time)

; General preferences

M564 S0 H0
G90                                            ; send absolute coordinates...
M83                                            ; ...but relative extruder moves
M550 P"Duet 3"                                 ; set printer name
M665 R221.628 L440.43 B184.9 H401.25           ; Set delta radius, diagonal rod length, printable radius and homed height
M666 X0 Y0 Z0                                 ; put your endstop adjustments here, or let auto calibration find them

M569 P0 S1                                     ; physical drive 0 goes backwards using default driver timings
M569 P1 S1                                     ; physical drive 1 goes backwards using default driver timings
M569 P2 S1                                     ; physical drive 2 goes backwards using default driver timings
M569 P3 S0                                     ; physical drive 3 goes forwards using default driver timings
M584 X0 Y1 Z2 E3                               ; set drive mapping
M350 X16 Y16 Z16 E16 I1                        ; configure microstepping with interpolation
M92 X160.00 Y160.00 Z160.00 E838.08                    ; set steps per mm
M566 X1200.00 Y1200.00 Z1200.00 E1500.00               ; set maximum instantaneous speed changes (mm/min)
M203 X18000.00 Y18000.00 Z18000.00 E1200.00            ; set maximum speeds (mm/min)
M201 X1000.00 Y1000.00 Z1000.00 E1000.00               ; set accelerations (mm/s^2)
M906 X1340 Y1340 Z1340 E1120 I30                       ; set motor currents (mA) and motor idle factor in per cent
M84 S30                                                ; Set idle timeout

M581 P"!e2stop" T2 C1

; Axis Limits
M208 Z-3 S1                                     ; set minimum Z

; Endstops
M574 X2 S1 P"xstop"                            ; configure active-high endstop for high end on X via pin xstop
M574 Y2 S1 P"ystop"                            ; configure active-high endstop for high end on Y via pin ystop
M574 Z2 S1 P"zstop"                            ; configure active-high endstop for high end on Z via pin zstop

M558 P8 H2 F1000 T12000 R0.2 S0.03 A5 C"servo0"

G31 P100 X0 Y0 Z-0.1

; Heaters2

M308 S0 P"bedtemp" Y"thermistor" T100000 B4300 C0 R4700

M950 H0 C"bed" T0                              ; create bed heater output (2.5) on bed and map it to sensor 0
M307 H0 B0 S1.00                               ; disable bang-bang mode for the bed heater and set PWM limit
M140 H0                                        ; map heated bed to heater 0
M143 H0 S120                                   ; set temperature limit for heater 0 to 120C
M308 S1 P"e0temp" Y"pt1000" R4700               ; configure sensor 1 as PT1000 on pin e0temp
M950 H1 C"e0heat" T1                           ; create nozzle heater output on e0heat and map it to sensor 1
M307 H1 B0 S1.00                               ; disable bang-bang mode for heater  and set PWM limit
M143 H1 S420                                   ; set temperature limit for heater 1 to 420C
; Heater model parameters

M307 H0 A26.8 C80.1 D0.6 S1.0 V0.0 B0
M307 H1 A261.3 C154.9 D2.0 S1.0 V0.0 B0

; Fans

M950 F0 C"fan0" Q500                                            ; Create fan 0 on pin fan0 and set its frequency
M106 P0 S0 H-1                                                    ; Set fan 0 value. Thermostatic control is turned off
M950 F1 C"fan1" Q500                                            ; Create fan 1 on pin fan1 and set its frequency
M106 P1 S1 H1 T45                                               ; Set fan 1 value. Thermostatic control is turned on
M950 F2 C"e1heat" Q500                                            ; Create fan 2 on pin fan2 and set its frequency
M106 P2 S1 H-1                                                    ; Set fan 2 value. Thermostatic control is turned off

; Tools
[... 59 lines cut ...]
    continue
  G30 P10 X34.64 Y20.00 Z-99999
  if result != 0
    continue
  G30 P11 X34.64 Y-20.00 Z-99999
  if result != 0
    continue
  G30 P12 X0.00 Y-40.00 Z-99999
  if result != 0
    continue
  G30 P13 X-34.64 Y-20.00 Z-99999
  if result != 0
    continue
  G30 P14 X-34.64 Y20.00 Z-99999
  if result != 0
    continue
  G30 P15 X0 Y0 Z-99999 S8
  if result != 0
    continue

  if move.calibration.final.deviation <= 0.04
    break
  echo "Repeating calibration because deviation is too high (" ^ move.calibration.final.deviation ^ "mm)" ; end loop
echo "Auto calibration successful, deviation", move.calibration.final.deviation ^ "mm"
G1 X0 Y0 Z150 F10000                ; get the head out of the way

jaysuk, 2020-12-15:
1. please review the contents of /var/log/syslog. This will give an indication if anything has failed on the pi side. How are you connecting the pi to the SKR?
2. have you looked on the console tab in DWC? this is where information is shown that it too long for a pop up. This is typical of G32.
3. you are missing the mesh information for G29. Try M557 R175 S16

sankafola, 2020-12-15:
Dec 15 00:00:14 duet3 rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="352" x-info="https://www.rsyslog.com"] rsyslogd was HUPed
Dec 15 00:00:14 duet3 systemd[1]: logrotate.service: Succeeded.
Dec 15 00:00:14 duet3 systemd[1]: Started Rotate log files.
Dec 15 00:00:24 duet3 systemd[1]: man-db.service: Succeeded.
Dec 15 00:00:24 duet3 systemd[1]: Started Daily man-db regeneration.
Dec 15 00:00:58 duet3 DuetControlServer[795]: [warn] Bad data checksum (expected 0x2d31, got 0x8068)
Dec 15 00:00:58 duet3 DuetControlServer[795]: [warn] Note: RepRapFirmware didn't receive valid data either (code 0x00000014)
Dec 15 00:00:58 duet3 DuetControlServer[795]: [warn] Bad data checksum (expected 0x2d31, got 0x0c4b)
Dec 15 00:00:58 duet3 DuetControlServer[795]: [warn] Note: RepRapFirmware didn't receive valid data either (code 0x00000000)
Dec 15 00:00:58 duet3 DuetControlServer[795]: [warn] Bad data checksum (expected 0x2d31, got 0xe439)
Dec 15 00:00:59 duet3 DuetControlServer[795]: [warn] Lost connection to Duet (Timeout while waiting for transfer ready pin)
Dec 15 00:03:43 duet3 systemd[1]: Starting Cleanup of Temporary Directories...
Dec 15 00:03:43 duet3 systemd[1]: systemd-tmpfiles-clean.service: Succeeded.
Dec 15 00:03:43 duet3 systemd[1]: Started Cleanup of Temporary Directories.
Dec 15 00:12:28 duet3 DuetWebServer[458]: #033[40m#033[32minfo#033[39m#033[22m#033[49m: Microsoft.AspNetCore.Hosting.Diagnostics[1]
Dec 15 00:12:28 duet3 DuetWebServer[458]:       Request starting HTTP/1.1 POST http://10.0.1.236/machine/code text/plain;charset=UTF-8 9
Dec 15 00:12:28 duet3 DuetWebServer[458]: #033[40m#033[32minfo#033[39m#033[22m#033[49m: Microsoft.AspNetCore.Routing.EndpointMiddleware[0]
Dec 15 00:12:28 duet3 DuetWebServer[458]:       Executing endpoint 'DuetWebServer.Controllers.MachineController.DoCode (DuetWebServer)'
Dec 15 00:12:28 duet3 DuetWebServer[458]: #033[40m#033[32minfo#033[39m#033[22m#033[49m: Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker[3]
Dec 15 00:12:28 duet3 DuetWebServer[458]:       Route matched with {action = "DoCode", controller = "Machine"}. Executing controller action with signature System.Threading.Tasks.Task`1[Microsoft.AspNetCore$
Dec 15 00:12:28 duet3 DuetWebServer[458]: #033[40m#033[32minfo#033[39m#033[22m#033[49m: DuetWebServer.Controllers.MachineController[0]
Dec 15 00:12:28 duet3 DuetWebServer[458]:       [DoCode] Executing code 'M112
Dec 15 00:12:28 duet3 DuetWebServer[458]:       M999'
Dec 15 00:17:01 duet3 CRON[6384]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Dec 15 00:27:55 duet3 DuetWebServer[458]: #033[40m#033[32minfo#033[39m#033[22m#033[49m: Microsoft.AspNetCore.Hosting.Diagnostics[1]
Dec 15 00:27:55 duet3 DuetWebServer[458]:       Request starting HTTP/1.1 GET http://10.0.1.236/Job/Status
Dec 15 00:27:55 duet3 DuetWebServer[458]: #033[40m#033[32minfo#033[39m#033[22m#033[49m: DuetWebServer.Middleware.CustomEndpointMiddleware[0]
Dec 15 00:27:55 duet3 DuetWebServer[458]:       No endpoint found for GET request via /Job/Status
Dec 15 00:27:55 duet3 DuetWebServer[458]: #033[40m#033[1m#033[33mwarn#033[39m#033[22m#033[49m: DuetWebServer.Middleware.FallbackMiddleware[0]
Dec 15 00:27:55 duet3 DuetWebServer[458]:       Could not find resource /Job/Status, serving index file
Dec 15 00:27:55 duet3 DuetWebServer[458]: #033[40m#033[32minfo#033[39m#033[22m#033[49m: Microsoft.AspNetCore.StaticFiles.StaticFileMiddleware[6]
Dec 15 00:27:55 duet3 DuetWebServer[458]:       The file /index.html was not modified
Dec 15 00:27:55 duet3 DuetWebServer[458]: #033[40m#033[32minfo#033[39m#033[22m#033[49m: Microsoft.AspNetCore.Hosting.Diagnostics[2]
Dec 15 00:27:55 duet3 DuetWebServer[458]:       Request finished in 17.1558ms 304 text/html
Dec 15 00:27:55 duet3 DuetWebServer[458]: #033[40m#033[32minfo#033[39m#033[22m#033[49m: DuetWebServer.Controllers.WebSocketController[0]
Dec 15 00:27:55 duet3 DuetWebServer[458]:       WebSocket disconnected from ::ffff:10.0.1.149:58120
Dec 15 00:27:55 duet3 DuetWebServer[458]: #033[40m#033[32minfo#033[39m#033[22m#033[49m: Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker[2]

gloomyandy, 2020-12-15:
@sankafola Please use our discord server (see: https://github.com/gloomyandy/RepRapFirmware/wiki/Support) to get support. It is much easier to handle these things over there.

LoicGRENON, 2020-12-22:
I have the same issue with this board and I still need to investigate but ... I noticed it's likely to happen when I interact with the printer while printing (the last time it's happened when the bed hits my arm ...)
I checked the syslog and DWC and there weren't any clues there.
I can't do it right now but soon I'll try to reproduce the behavior explained above to see if it's related to and will be back !

gloomyandy, 2020-12-24:
@LoicGRENON Please use our discord server for support.

closed 2021-01-21 by gloomyandy