gloomyandy/RepRapFirmware #61: Sensorless homing skr 1.4 + tmc2209 not working

tobyworks, 2020-12-19:
Hi i followed this guide step by step:

https://www.youtube.com/watch?v=s68U6BwxHpI&t=1472s

also checkout out the documentation on sensorless homing, but it never triggers. I have connected the TMC2209 to skr 1.4 board like shown here:

[image]

But still no luck when running:

M574 X1 S3 ; configure endstop for stall detection
M915 P0 S-127 H10 R1 ; Set very sensitive stall detect
G1 X200 H1 ; Should stall pretty much immediately

Is there something I am missing here? It's not a faulty driver since i've 3 seperate TMC2209's

gloomyandy, 2020-12-19:
Please use the discord server for support. See the Wiki for how to find the server.

tobyworks, 2020-12-20:
will do!

closed 2020-12-20 by tobyworks