gloomyandy/RepRapFirmware #65: have tried everything

bl00d0und42, 2021-01-18:
**can someone help me?

have a SKR1.4 and a
ESP8266.

wired as follows:
https://www.jayuk.org/compiling-and-running-reprapfirmware-on-an-skr-1-4-and-other-lp17xx-boards/

have the "Duet WiFi Socket Server for LPC1768 and STM32F4 versions of RRF // DuetWiFiServer-lpc.bin
"installed on the ESP and the" RepRapFirmware 3.2 Port for LPC and STM32F4 (v3.2_1) // firmware-lpc-esp8266wifi-3.2_1.bin
"on the SKR.

I can operate the printer via Octoprint, the ESP connects to the WLAN (here I had to manually assign the IP). if I now want to surf the IP of the ESP, only NO_RESPONSE or CONNECTION_REFUSED comes up.**

wolfgangmauer, 2021-01-18:
For me a M588 S"*" helps a lot (remove all known networks)

bl00d0und42, 2021-01-18:
doesnt work :-(

jaysuk, 2021-01-18:
can you join us on discord for support please.
https://discord.gg/uS97Qs7

closed 2021-01-18 by gloomyandy