gloomyandy/RepRapFirmware #68: Telnet issues

The-Moops, 2021-02-08:
I can't seem to connect to my LPC board through Telnet, I've enabled it through M586 and it looks like it tries to connect but instantly fails. Is there some config step I'm missing?

jaysuk, 2021-02-08:
Telnet isn't supported on the LPC builds

https://teamgloomy.github.io/lpc_differences.html

Thanks,

Jay

The-Moops, 2021-02-08:
Thanks, not sure how I didn't catch that. I'm assuming no plans for future support given the lack of RAM?

jaysuk, 2021-02-08:
Yep, unfortunately there are no plans to activate Telnet support.
Ram and the lack of concurrent connections are the main issues.

The STM32 builds do not have this restriction

Thanks,

Jay

gloomyandy, 2021-02-08:
Note that Telnet is not currently supported on the STM32F4 versions either, I've never really seen the need.

closed 2021-02-09 by The-Moops