gloomyandy/RepRapFirmware #70: esp M587/588 failed

whoim2, 2021-02-17:
Hello. skr1.3 dip, Lolin ESP8266MOD with ch340.
RepRapFirmware for LPC176x based Boards (biquskr_1.3) version 3.2_4 running on LPC176x at 100Mhz
Used output buffers: 1 of 16 (2 max)
this (https://github.com/gloomyandy/DuetWiFiSocketServer/releases/tag/V1.25-01) FW esp8266 (try -el and without -el).

M552 S0 ok
M588 S"*" Error: M588: Failed to reset the WiFi module to factory settings: another SPI transfer is pending
ok
M587 S"Nika" P"4347544C" Error: M587: Failed to add SSID to remembered list: another SPI transfer is pending

help please

whoim2, 2021-02-17:
> === Diagnostics ===
> RepRapFirmware for LPC176x based Boards (biquskr_1.3) version 3.2_4 running on LPC176x at 100Mhz
> Used output buffers: 1 of 16 (2 max)
> === RTOS ===
> Static ram: 4956
> Dynamic Memory (RTOS Heap 5): 8136 free, 8136 never used
> Allocations: 168 Frees: 13
> Never used RAM 256, free system stack 66 words
> Tasks: NETWORK(ready,495) HEAT(blocked,249) MAIN(running,358) IDLE(ready,21)
> Owned mutexes: USB(MAIN)
> === Platform ===
> Last reset 00:00:56 ago, cause: [power up][reset button]
> Last software reset time unknown, reason: HardFault imprec, GCodes spinning, available RAM 256, slot 1
> Software reset code 0x0063 HFSR 0x40000000 CFSR 0x00000400 ICSR 0x0042a803 BFAR 0xe000ed38 SP 0x2007d068 Task MAIN Freestk 364 ok
> Stack: 50004180 8400000c 000000cd 09478048 00000001 00051eff 00051f06 01000000 0000000c 0000000c 10000c6c 10002ed8 00000000 0004b67f 0004b6dd 00000000 10002ed8 10000c68 00000000 0004ba8b 10002edc 00000001 00000000 2007d178 2007d864 0000024c 00000000
> Error status: 0x00
> Aux0 errors 0,0,0
> Supply voltage: under voltage events: 0
> Driver 0: position 66237, standstill, SG min/max 0/0, error r/w 0/0, ifcnt 12, cnt r/w 4906/12, timeout 0, failedOp 0xff
> Driver 1: position 66237, standstill, SG min/max 0/0, error r/w 0/0, ifcnt 12, cnt r/w 4906/12, timeout 0, failedOp 0xff
> Driver 2: position 66237, standstill, SG min/max 0/0, error r/w 0/0, ifcnt 12, cnt r/w 4906/12, timeout 0, failedOp 0xff
> Driver 3: position 0, standstill, SG min/max 0/0, error r/w 0/0, ifcnt 12, cnt r/w 4905/12, timeout 0, failedOp 0xff
> Driver 4: position 0
> Driver 5: position 0
> Driver 6: position 0
> Date/time: 1970-01-01 00:00:00
> Slowest loop: 99.85ms; fastest: 0.13ms
> Step timer: target 0 count 56787076 delta -56787077 late 0
> USBSerial connected 1
> ADC not ready 2 ADC error threshold 10 ADC Init 1
> Ints: 0; Calls 0; fast: 4294967295uS; slow 0uS adj 0 bad 0 big delta 0
> PWM Channels
> Delta -1444471863 Start -1 End -1
> === Storage ===
> Free file entries: 6
> SD card 0 detected
> SD card longest read time 1.7ms, write time 0.0ms, max retries 0
> === Move ===
> DMs created 40, maxWait 0ms, bed compensation in use: none, comp offset 0.000
> === DDARing ===
> Scheduled moves 0, completed moves 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
> === Heat ===
> Bed heaters = 0, chamberHeaters = -1
> === GCodes ===
> Segments left: 0
> Movement lock held by null
> HTTP is idle in state(s) 0
> File is idle in state(s) 0
> USB is ready with "M122" in state(s) 0
> Aux is idle in state(s) 0
> Trigger is idle in state(s) 0
> Queue is idle in state(s) 0
> Daemon is idle in state(s) 0
> Autopause is idle in state(s) 0
> Code queue is empty.
> === Network ===
> Slowest loop: 0.96ms; fastest: 0.00ms
> Responder states: HTTP(0) HTTP(0)
> HTTP sessions: 0 of 2
> - WiFi -
> Network state is starting2
> WiFi module is disabled
> Failed messages: pending 0, notready 0, noresp 0
> Socket states: 0 0

jaysuk, 2021-02-17:
Its a wiring issue as your wifi state never changes to idle

whoim2, 2021-02-18:
> Its a wiring issue as your wifi state never changes to idle
> […](#)

im test my wiring.. maybe, long wires? About 30cm, from tft24 display (exp1/exp2), with smd resistors
[image]

whoim2, 2021-02-18:
[image]

gloomyandy, 2021-02-18:
Please use the discord channels for support: https://teamgloomy.github.io/support.html

whoim2, 2021-02-25:
Ok, thanks.
Problem in flashing esp, im use flash downloader. After flashing from esptools.py work perfectly. Thanks!

closed 2021-02-25 by whoim2