gloomyandy/RepRapFirmware #74: Pi3 duetcontrolserver install error

wolfgangmauer, 2021-03-11:
I try a clean install with DuetPi-lite and get the error
Executing ./RRF_LPC_3_2_RC_2.sh
......
Setting up duetwebcontrol (3.3-b2) ...
Setting up duetsd (1.1.0) ...
Setting up duetcontrolserver (3.3-b2) ...
Installing new version of config file /opt/dsf/conf/config.json ...
dpkg: error processing package duetcontrolserver (--configure):
 installed duetcontrolserver package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of duetwebserver:
 duetwebserver depends on duetcontrolserver (>= 3.3-b2); however:
  Package duetcontrolserver is not configured yet.

dpkg: error processing package duetwebserver (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of reprapfirmware:
 reprapfirmware depends on duetcontrolserver (>= 1.2.0.0); however:
  Package duetcontrolserver is not configured yet.

dpkg: error processing package reprapfirmware (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of duetpluginservice:
 duetpluginservice depends on duetcontrolserver (= 3.3-b2); however:
  Package duetcontrolserver is not configured yet.

dpkg: error processing package duetpluginservice (--configure):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for systemd (241-7~deb10u6+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for dbus (1.12.20-0+deb10u1) ...
Errors were encountered while processing:
 duetcontrolserver
 duetwebserver
 reprapfirmware
 duetpluginservice
E: Sub-process /usr/bin/dpkg returned an error code (1)

jaysuk, 2021-03-11:
If the updates fail with an error exit status 1, run the following and then
rerun the script.

sudo mv /var/lib/dpkg/info/duet* /tmp/
sudo mv /var/lib/dpkg/info/reprapfirmware.* /tmp/
sudo apt-get update
sudo dpkg --configure -a
sudo apt-get install --reinstall duetcontrolserver

Thanks,

Jay

wolfgangmauer, 2021-03-11:
Sorry, but didn't help

pi@duet3:~ $ sudo mv /var/lib/dpkg/info/duet* /tmp/
pi@duet3:~ $ sudo mv /var/lib/dpkg/info/reprapfirmware.* /tmp/
pi@duet3:~ $ sudo apt-get update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:3 https://pkg.duet3d.com unstable InRelease
Reading package lists... Done
pi@duet3:~ $ sudo dpkg --configure -a
Setting up duetcontrolserver (3.2.0) ...
Setting up duettools (3.2.0) ...
Setting up duetwebserver (3.2.0) ...
Setting up reprapfirmware (3.2.0-1) ...
Setting up duetsoftwareframework (3.2.0) ...
Setting up duetwebcontrol (3.2.0) ...
pi@duet3:~ $ sudo apt-get install --reinstall duetcontrolserver
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  duetruntime duetsoftwareframework duettools duetwebcontrol duetwebserver reprapfirmware
Recommended packages:
  duetpluginservice
The following packages will be upgraded:
  duetcontrolserver duetruntime duetsoftwareframework duettools duetwebcontrol duetwebserver reprapfirmware
7 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 33.0 MB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 https://pkg.duet3d.com unstable/armv7 armhf duetsoftwareframework armhf 3.3-b2 [2,046 B]
Get:2 https://pkg.duet3d.com unstable/armv7 armhf duettools armhf 3.3-b2 [55.0 kB]
Get:3 https://pkg.duet3d.com unstable/armv7 armhf duetcontrolserver armhf 3.3-b2 [227 kB]
Get:4 https://pkg.duet3d.com unstable/armv7 armhf duetruntime armhf 3.3-b2 [26.9 MB]
Get:5 https://pkg.duet3d.com unstable/armv7 armhf duetwebserver armhf 3.3-b2 [80.2 kB]
Get:6 https://pkg.duet3d.com unstable/armv7 armhf duetwebcontrol all 3.3-b2 [4,762 kB]
Get:7 https://pkg.duet3d.com unstable/armv7 armhf reprapfirmware all 3.3-b2-1 [1,000 kB]
Fetched 33.0 MB in 15s (2,188 kB/s)
Reading changelogs... Done
dpkg: warning: files list file for package 'duetruntime' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'duettools' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'duetwebserver' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'duetsoftwareframework' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'reprapfirmware' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'duetwebcontrol' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'duetsd' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'duetcontrolserver' missing; assuming package has no files currently installed
(Reading database ... 57971 files and directories currently installed.)
Preparing to unpack .../0-duetsoftwareframework_3.3-b2_armhf.deb ...
Unpacking duetsoftwareframework (3.3-b2) over (3.2.0) ...
Preparing to unpack .../1-duettools_3.3-b2_armhf.deb ...
Unpacking duettools (3.3-b2) over (3.2.0) ...
Preparing to unpack .../2-duetcontrolserver_3.3-b2_armhf.deb ...
Unpacking duetcontrolserver (3.3-b2) over (3.2.0) ...
Preparing to unpack .../3-duetruntime_3.3-b2_armhf.deb ...
Unpacking duetruntime (3.3-b2) over (3.2.0) ...
Preparing to unpack .../4-duetwebserver_3.3-b2_armhf.deb ...
Unpacking duetwebserver (3.3-b2) over (3.2.0) ...
Preparing to unpack .../5-duetwebcontrol_3.3-b2_all.deb ...
Unpacking duetwebcontrol (3.3-b2) over (3.2.0) ...
Preparing to unpack .../6-reprapfirmware_3.3-b2-1_all.deb ...
Unpacking reprapfirmware (3.3-b2-1) over (3.2.0-1) ...
Setting up duetruntime (3.3-b2) ...
Setting up duetcontrolserver (3.3-b2) ...

Configuration file '/opt/dsf/conf/config.json'
 ==> File on system created by you or by a script.
 ==> File also in package provided by package maintainer.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
[... 27 lines cut ...]
dpkg: error processing package duetsoftwareframework (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of reprapfirmware:
 reprapfirmware depends on duetsoftwareframework (>= 3.2.0); however:
  Package duetsoftwareframework is not configured yet.
 reprapfirmware depends on duetcontrolserver (>= 1.2.0.0); however:
  Package duetcontrolserver is not configured yet.

dpkg: error processing package reprapfirmware (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of duetwebcontrol:
 duetwebcontrol depends on duetsoftwareframework (>= 3.2.0); however:
  Package duetsoftwareframework is not configured yet.

dpkg: error processing package duetwebcontrol (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 duetcontrolserver
 duettools
 duetwebserver
 duetsoftwareframework
 reprapfirmware
 duetwebcontrol
E: Sub-process /usr/bin/dpkg returned an error code (1)
pi@duet3:~ $

wolfgangmauer, 2021-03-13:
I think a have a broken lpcTfrReadyPin :-(

closed 2021-03-13 by wolfgangmauer