gloomyandy/RepRapFirmware #76: [FR] MKS NANO V3

Lino77, 2021-03-31:
IS ist Possible to add the MKS NANO V3,

gloomyandy, 2021-04-01:
It may be possible, but MKS have chosen to use SPI3 for the SD card interface (other boards use either SPI1 or SDIO). I don't have access to the board and so have no way to test it and given that it does not seem to offer anything over other boards that we already support I don't think supporting it will be very high on my list.

Lino77, 2021-04-01:
okay, then i'm assuming that my idea of using the wifi module is also wrong
[image]

gloomyandy, 2021-04-02:
"okay, then i'm assuming that my idea of using the wifi module is also wrong"

I'm not really sure what you mean by this comment, could you explain? There is a reasonable chance that the WiFi adaptor will work (I haven't looked in detail, but it is using SPI2 which is what other modules use), but I've not looked at it in detail. However for anything to work you need to get the SD card interface working, and on this board it is using an SPI device that we do not normally use. The latest version of the firmware does support the use of SPI3, but it has not been tested.

Lino77, 2021-04-02:
i have the board here and the wifi module, and thought to myself that it is a stm32, it would also work on it. I wrote a few files firmware compiled but the board does not boot. I think there is still something missing. sd det pin ?
my two files
[mksreprap.zip](https://github.com/gloomyandy/RepRapFirmware/files/6250573/mksreprap.zip)

WonderRat, 2021-04-02:
Try to connect t your board and get debug message with signature (look in BoardConfig.cpp).

>    {PA_7, "PA7"}, //IO4 LPC Data Ready
PC7 in scheme

>    {PC_2, "PB14"},//MISOSPI2
>    {PC_3, "PB15"},//MOSISPI2

>    {   //CLK, MISO, MOSI
>        {PA_5, PA_6, PB_5},                     //SPI0
MOSI PA_7

>        {NoPin, NoPin, NoPin},                  //SPI3
TMC SPI pins?

I don't have the board.

Lino77, 2021-04-02:
Oh, typo
{PC_2, "PC2"}, // MISOSPI2
{PC_3, "PC3"}, // MOSISPI2
is correct
unfortunately does not come via usb the board does not boot
[image]

WonderRat, 2021-04-02:
May be it's because firmware expecting to be loaded to address 0x8008000, but this board uses 0x0800C000 (it's from Marlin's platformio.ini) ?

Lino77, 2021-04-04:
okay if that's really the case, then a new linker script is needed for the mks, which is beyond my knowledge.
then I have to wait until my fly board comes. to keep playing

WonderRat, 2021-04-04:
Stupid idea: cut out 0x4000 of zeros from beginning of binary. If there is no reading from 0x8008000-0x800ffff then it may work.
Here is binary if you don't afraid. I don't know what i'm doing >:->
Don't try it if you can't restore firmware by yourself (via ST-Link or something).

[firmware-stm32f4-esp8266wifi-3.3beta1_5_test_0x800c000.bin.zip](https://github.com/gloomyandy/RepRapFirmware/files/6254370/firmware-stm32f4-esp8266wifi-3.3beta1_5_test_0x800c000.bin.zip)

Lino77, 2021-04-04:
@WonderRat hat nicht funktioniert, ich warte bis gloomyandy Zeit findet sich das mal anzuschauen

closed 2021-04-04 by Lino77

WonderRat, 2021-04-25:
Try replace memory areas section in ldscript:
```
/* Specify the memory areas */
MEMORY
{
VECTOR (rx)    : ORIGIN = 0x800c000,  LENGTH = 16K
RESETDATA (rx) : ORIGIN = 0x8010000,  LENGTH = 64K
FLASH (rx)     : ORIGIN = 0x8020000,  LENGTH = 1024K - 128K
RAM (xrw)      : ORIGIN = 0x20000000, LENGTH = 128K
CCMRAM (rw)    : ORIGIN = 0x10000000, LENGTH = 64K
}
```

And another stupid idea: get rid of bootloader. Duplicate first 0x8000 bytes of standard firmware.bin (just for vectors table) and flash it at 0x8000000 using ST-LINK. Its only for checking firmware booting and response through USB.
[firmware-stm32f4-esp8266wifi-3.3beta1_6-kill_bootloader.bin.zip](https://github.com/gloomyandy/RepRapFirmware/files/6371315/firmware-stm32f4-esp8266wifi-3.3beta1_6-kill_bootloader.bin.zip)

Lino77, 2021-05-01:
@WonderRat  https://discord.gg/6GbqcVhS

WonderRat, 2021-05-02:
Fucking chat? No.

Lino77, 2021-05-02:
@WonderRat I have the bootloder source from the skr2

WonderRat, 2021-05-03:
Now rewrite the work with SD from SDIO to SPI3 :-D
And don't forget to check the use of heater and fan pins.

WonderRat, 2021-05-08:
So, how it's going? Trick with removing bootloader works?

https://github.com/feaser/openblt/releases/tag/openblt_v011101
https://www.feaser.com/openblt/doku.php?id=homepage
Open source bootloader with samples:
ARMCM4_STM32F4_Olimex_STM32P405_GCC - flashes firmware from SD connected to SPI2 (mmc.c)
ARMCM3_STM32F2_Olimex_STM32P207_GCC - SPI3

I adapted the example for my BluePill.
I copied file.c to Boot folder and removed s-record parsing from there and added flash_layout.c.
[OpenBLT_BluePill_SPI2.zip](https://github.com/gloomyandy/RepRapFirmware/files/6445723/OpenBLT_BluePill_SPI2.zip)
It flashes firmware.bin to 0x8008000 from SD connected to SPI2. I tried change to SPI1 but failed.

Lino77, 2021-05-10:
The problem is not with the bootloader. klipper runs on it and so does Marlin. with klipper only one line was inserted in the code. 48 K bootloader
https://github.com/KevinOConnor/klipper/commit/c4edc7d946fd955a1d94dd888244bf11cd3b6326#diff-1b2b9f5776a824bd8dac8ccc88e17717672dd76aa8a8ad0f51ba816a323dcb36

WonderRat, 2021-05-10:
You don’t telling the results of the experiments, so it’s hard to tell what the problem is.

When you tried modified ldscript (what inside?) you executed "make clean" to delete temporary files from previous attempts?

Did you tried 32KB bootloader with standard RRF builds?

I have compiled the bootloader but haven't tested it (I don't have STM32F4). It's just a launcher for a test. What is the frequency of the quartz resonator? I wrote in blt_conf.h 8MHz.

[MKS_Robin_Nano_V3_bootloader_test_32KB_no_fw_update.zip](https://github.com/gloomyandy/RepRapFirmware/files/6453982/MKS_Robin_Nano_V3_bootloader_test_32KB_no_fw_update.zip)

Lino77, 2021-05-13:
@WonderRat neither work. I have the source from the nano V3 bootloader, but I can't do anything with it. it would be easier to shorten it to 32 K for someone who knows what to do with it.

gloomyandy, 2021-05-13:
Can I suggest that if you are modifying/creating/testing a bootloader, you create a new gitgub repo for that code and continue the discussion of it there.

WonderRat, 2021-05-13:
Yeah, just publish that sources - may be someone will help. 
BTW, make a whole flash memory dump (1MB) startingfrom 0x8000000 to check if everything is in place.