gloomyandy/RepRapFirmware #77: LPC1768 sgen_l voltage detection support

billhu1996, 2021-04-01:
Is it possible for LPC motherboards to use voltage detection, even with some hardware additions/changes? I saw on this [page](https://duet3d.dozuki.com/Wiki/Setting_up_to_resume_a_print_after_a_power_failure) that it is necessary to have voltage detection data to resume printing after a power failure.

billhu1996, 2021-04-01:
Thank you for your answer. This is indeed a good way.

But I have many suitable capacitors for storing electricity. If the motherboard can perform voltage detection (for example, through the E1 A/D pin or servo A/D pin or other A/D pins, and then through a suitable resistor in series), there is no need to purchase a UPS module.

I found that the STM32 motherboard already supports voltage detection, is it possible that the LPC motherboard also supports it?

jaysuk, 2021-04-01:
The code is STM32 only and there are no plans to add it to the LPC builds,
especially as none of the LPC boards natively support it.

billhu1996, 2021-04-02:
Thanks for your answer

closed 2021-04-02 by billhu1996