gloomyandy/RepRapFirmware #79: not compiling for STM32

dineshvyas, 2021-04-05:
VSCode gives Error. possible reason absence of stdbool.h file in repo. screenshot included. 
[image]
[image]
any direction?

gloomyandy, 2021-04-05:
No real idea, quite a few people have been able to build from source so it is almost certainly a problem with your configuration. It looks like you are getting errors in the makefiles (Ignore the problems reported by vscode, I've made no effort to clean any of those up), so I'd start by looking there. What version of make do you have installed?

The new Wiki does not include information of building from source, but you may find the old one useful...
https://github.com/gloomyandy/RepRapFirmware/wiki/build-instructions

jaysuk, 2021-04-05:
There's this page on the new wiki
https://teamgloomy.github.io/firmware_build.html

dineshvyas, 2021-04-06:
I followed steps as described in page mentioned https://github.com/gloomyandy/RepRapFirmware/wiki/build-instructions.
my make version is 3.81.2525.1772.
[image]

gloomyandy, 2021-04-06:
Ignore the vscode "problems" tab, what you need to do is concentrate on getting make to actually build the project. What errors are you getting from the make process. Many others have been able to build the project, so something is wrong with how you have things setup. Have you made any changes to the makefiles (or any other files).

dineshvyas, 2021-04-07:
I haven't changed anything in anyfile. However i will continue to dig it. Thanks for all the help. Kindly keep this ticket open.

WonderRat, 2021-04-08:
Your compiler must have stdbool.h. Check environment variables, paths for includes.
If nothing helps you can google that file and copypaste it, but you can get errors about missing other std*.h

dineshvyas, 2021-04-11:
> Your compiler must have stdbool.h. Check environment variables, paths for includes.
> If nothing helps you can google that file and copypaste it, but you can get errors about missing other std*.h

Yes exactly after copying stdbool.h it give error for more missing .h files. however I installed make version 4.3 and it didn't change anything. platformIO could an option here.

jaysuk, 2021-04-11:
no issues when compiling here. 
Just taken a snapshot of all the required repositories, dropped them into vscode in the appropriate folders and its compiled fine.
no changes or dependencies required

closed 2021-04-22 by gloomyandy