gloomyandy/RepRapFirmware #80: Error Processing When Configuring DuetPi Lite

cjsimon2, 2021-04-12:
I've tried multiple times with no luck:
pi@duet3:~ $ sudo chmod 755 RRF_STM_3_2.sh
pi@duet3:~ $ ./RRF_STM_3_2.sh
-----This will install the STM Version of DSF-----
-----Update and upgrade the PI-----
duetsoftwareframework set on hold.
duetcontrolserver set on hold.
duetruntime set on hold.
duetsd set on hold.
duetsoftwareframework set on hold.
duettools set on hold.
duetwebcontrol set on hold.
duetwebserver set on hold.
reprapfirmware set on hold.
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Hit:3 https://pkg.duet3d.com stable InRelease
Reading package lists...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
7 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
W: APT had planned for dpkg to do more than it reported back (0 vs 15).
   Affected packages: duetcontrolserver:armhf duetsd:armhf duetsoftwareframework:armhf duettools:armhf duetwebcontrol:armhf duetwebserver:armhf reprapfirmware:armhf
-----Upgrade and Update finished-----
-----Swichting to the unstable branch-----
-----Switching finished-----
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Get:3 https://pkg.duet3d.com unstable InRelease [2,565 B]
Get:4 https://pkg.duet3d.com unstable/armv7 armhf Packages [53.3 kB]
Fetched 55.9 kB in 2s (27.8 kB/s)
Reading package lists...
Building dependency tree...
Reading state information...
7 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
duetsd is already the newest version (1.1.0).
duetsd set to manually installed.
The following held packages will be changed:
  duetcontrolserver duetruntime duetsoftwareframework duettools duetwebcontrol duetwebserver reprapfirmware
The following packages will be DOWNGRADED:
  duetcontrolserver duetruntime duetsoftwareframework duettools duetwebcontrol duetwebserver reprapfirmware
0 upgraded, 0 newly installed, 7 downgraded, 0 to remove and 0 not upgraded.
7 not fully installed or removed.
Need to get 34.9 MB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://pkg.duet3d.com unstable/armv7 armhf duetsoftwareframework armhf 3.2.0 [2,022 B]
Get:2 https://pkg.duet3d.com unstable/armv7 armhf duettools armhf 3.2.0 [62.3 kB]
Get:3 https://pkg.duet3d.com unstable/armv7 armhf duetwebcontrol all 3.2.0 [4,704 kB]
Get:4 https://pkg.duet3d.com unstable/armv7 armhf reprapfirmware all 3.2.0-1 [958 kB]
Get:5 https://pkg.duet3d.com unstable/armv7 armhf duetwebserver armhf 3.2.0 [85.1 kB]
Get:6 https://pkg.duet3d.com unstable/armv7 armhf duetcontrolserver armhf 3.2.0 [230 kB]
Get:7 https://pkg.duet3d.com unstable/armv7 armhf duetruntime armhf 3.2.0 [28.8 MB]
Fetched 34.9 MB in 15s (2,396 kB/s)                                                                                                                                
dpkg: warning: downgrading duetsoftwareframework from 3.2.2 to 3.2.0
(Reading database ... 60327 files and directories currently installed.)
Preparing to unpack .../0-duetsoftwareframework_3.2.0_armhf.deb ...
Unpacking duetsoftwareframework (3.2.0) over (3.2.2) ...
To delete the "dsf" user run as root: userdel --force dsf
dpkg: warning: downgrading duettools from 3.2.2 to 3.2.0
Preparing to unpack .../1-duettools_3.2.0_armhf.deb ...
Unpacking duettools (3.2.0) over (3.2.2) ...
dpkg: warning: downgrading duetwebcontrol from 3.2.2 to 3.2.0
Preparing to unpack .../2-duetwebcontrol_3.2.0_all.deb ...
Unpacking duetwebcontrol (3.2.0) over (3.2.2) ...
dpkg: warning: downgrading reprapfirmware from 3.2.2-1 to 3.2.0-1
Preparing to unpack .../3-reprapfirmware_3.2.0-1_all.deb ...
Unpacking reprapfirmware (3.2.0-1) over (3.2.2-1) ...
dpkg: warning: downgrading duetwebserver from 3.2.2 to 3.2.0
Preparing to unpack .../4-duetwebserver_3.2.0_armhf.deb ...
Unpacking duetwebserver (3.2.0) over (3.2.2) ...
dpkg: warning: downgrading duetcontrolserver from 3.2.2 to 3.2.0
Preparing to unpack .../5-duetcontrolserver_3.2.0_armhf.deb ...
Unpacking duetcontrolserver (3.2.0) over (3.2.2) ...
To delete "gpio" group run as root: groupdel gpio
dpkg: warning: downgrading duetruntime from 3.2.2 to 3.2.0
Preparing to unpack .../6-duetruntime_3.2.0_armhf.deb ...
Unpacking duetruntime (3.2.0) over (3.2.2) ...
Setting up duetruntime (3.2.0) ...
dpkg: dependency problems prevent configuration of duetsoftwareframework:
 duetsoftwareframework depends on duetsd (= 1.1.0); however:
  Package duetsd is not configured yet.

dpkg: error processing package duetsoftwareframework (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of reprapfirmware:
 reprapfirmware depends on duetsoftwareframework (>= 3.2.0); however:
  Package duetsoftwareframework is not configured yet.
 reprapfirmware depends on duetsd (>= 1.0.5); however:
  Package duetsd is not configured yet.

dpkg: error processing package reprapfirmware (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of duetwebcontrol:
 duetwebcontrol depends on duetsoftwareframework (>= 3.2.0); however:
  Package duetsoftwareframework is not configured yet.

dpkg: error processing package duetwebcontrol (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of duetcontrolserver:
 duetcontrolserver depends on duetsoftwareframework (>= 3.2.0); however:
  Package duetsoftwareframework is not configured yet.

dpkg: error processing package duetcontrolserver (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of duettools:
 duettools depends on duetcontrolserver (= 3.2.0); however:
  Package duetcontrolserver is not configured yet.

dpkg: error processing package duettools (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of duetwebserver:
 duetwebserver depends on duetcontrolserver (>= 3.2.0); however:
  Package duetcontrolserver is not configured yet.

dpkg: error processing package duetwebserver (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 duetsoftwareframework
 reprapfirmware
 duetwebcontrol
 duetcontrolserver
 duettools
 duetwebserver
E: Sub-process /usr/bin/dpkg returned an error code (1)
pi@duet3:~ $

jaysuk, 2021-04-12:
try running

sudo mv /var/lib/dpkg/info/duet* /tmp/
sudo mv /var/lib/dpkg/info/reprapfirmware.* /tmp/
sudo apt-get update
sudo dpkg --configure -a
sudo apt-get install --reinstall duetcontrolserver

Thanks,

Jay

jaysuk, 2021-04-12:
can you try using this script?

https://raw.githubusercontent.com/TeamGloomy/LPC-STM32-DSF-Install_Script/master/RRF_STM_3_2_2_test.sh

cjsimon2, 2021-04-13:
Hi @jaysuk 
Thanks for the help. I don't think it worked....but I did not get an error.  Still failing to connect on Duet web control.

Here is what I got on the install:
Linux duet3 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@duet3:~ $ wget https://raw.githubusercontent.com/TeamGloomy/LPC-STM32-DSF-Install_Script/master/RRF_STM_3_2_2_test.sh
--2021-04-13 10:03:23--  https://raw.githubusercontent.com/TeamGloomy/LPC-STM32-DSF-Install_Script/master/RRF_STM_3_2_2_test.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.111.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1477 (1.4K) [text/plain]
Saving to: ‘RRF_STM_3_2_2_test.sh’

RRF_STM_3_2_2_test.sh                    100%[==================================================================================>]   1.44K  --.-KB/s    in 0.003s  

2021-04-13 10:03:23 (536 KB/s) - ‘RRF_STM_3_2_2_test.sh’ saved [1477/1477]

pi@duet3:~ $ sudo chmod 755 RRF_STM_3_2_2_test.sh
pi@duet3:~ $ ./RRF_STM_3_2_2_test.sh
-----This will install the STM Version of DSF-----
-----Update and upgrade the PI-----
duetsoftwareframework set on hold.
duetcontrolserver set on hold.
duetruntime set on hold.
duetsd set on hold.
duettools set on hold.
duetwebcontrol set on hold.
duetwebserver set on hold.
reprapfirmware set on hold.
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.9 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [372 kB]
Hit:4 https://pkg.duet3d.com stable InRelease
Get:5 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Fetched 13.4 MB in 10s (1,320 kB/s)
Reading package lists...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  alsa-utils avahi-daemon base-files bind9-host curl groff-base iputils-ping libavahi-client3 libavahi-common-data libavahi-common3 libavahi-core7 libbind9-161
  libbsd0 libcurl4 libdns-export1104 libdns1104 libisc-export1100 libisc1100 libisccc161 libisccfg163 libldap-2.4-2 libldap-common liblwres161 libopenjp2-7
  libpam-systemd libpython3.7 libpython3.7-minimal libpython3.7-stdlib libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 libssl1.1
  libsystemd0 libtiff5 libudev1 libzstd1 openssh-client openssh-server openssh-sftp-server openssl python3.7 python3.7-minimal raspberrypi-bootloader
  raspberrypi-kernel raspberrypi-sys-mods rpi-eeprom ssh systemd systemd-sysv udev
51 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 138 MB of archives.
After this operation, 3,858 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf libssl1.1 armhf 1.1.1d-0+deb10u6+rpt1 [1,277 kB]
Get:2 http://mirror.os6.org/raspbian/raspbian buster/main armhf base-files armhf 10.3+rpi1+deb10u9 [70.1 kB]
Get:3 http://mirrors.ocf.berkeley.edu/raspbian/raspbian buster/main armhf systemd-sysv armhf 241-7~deb10u7+rpi1 [100 kB]
Get:4 http://mirrors.ocf.berkeley.edu/raspbian/raspbian buster/main armhf libpam-systemd armhf 241-7~deb10u7+rpi1 [194 kB]
Get:5 http://mirrors.ocf.berkeley.edu/raspbian/raspbian buster/main armhf libsystemd0 armhf 241-7~deb10u7+rpi1 [306 kB]
Get:6 http://mirrors.ocf.berkeley.edu/raspbian/raspbian buster/main armhf systemd armhf 241-7~deb10u7+rpi1 [3,252 kB]          
Get:7 http://archive.raspberrypi.org/debian buster/main armhf openssh-sftp-server armhf 1:7.9p1-10+deb10u2+rpt1 [36.0 kB]  
Get:8 http://archive.raspberrypi.org/debian buster/main armhf openssh-server armhf 1:7.9p1-10+deb10u2+rpt1 [291 kB]
Get:10 http://archive.raspberrypi.org/debian buster/main armhf openssh-client armhf 1:7.9p1-10+deb10u2+rpt1 [669 kB]
Get:12 http://mirror.os6.org/raspbian/raspbian buster/main armhf libpython3.7 armhf 3.7.3-2+deb10u3 [1,253 kB]
[... 1116 lines cut ...]
 kernel:[  856.972443]  r4:c31d75c0

Message from syslogd@duet3 at Apr 13 10:10:38 ...
 kernel:[  856.972454] [<c02456cc>] (kthread) from [<c02000ec>] (ret_from_fork+0x14/0x28)

Message from syslogd@duet3 at Apr 13 10:10:38 ...
 kernel:[  856.972465] Exception stack(0xc3237fb0 to 0xc3237ff8)

Message from syslogd@duet3 at Apr 13 10:10:38 ...
 kernel:[  856.972474] 7fa0:                                     00000000 00000000 00000000 00000000

Message from syslogd@duet3 at Apr 13 10:10:38 ...
 kernel:[  856.972486] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

Message from syslogd@duet3 at Apr 13 10:10:38 ...
 kernel:[  856.972498] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000

Message from syslogd@duet3 at Apr 13 10:10:38 ...
 kernel:[  856.972509]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c02456cc

Message from syslogd@duet3 at Apr 13 10:10:38 ...
 kernel:[  856.972519]  r4:c30d9580

Message from syslogd@duet3 at Apr 13 10:10:38 ...
 kernel:[  856.972529] Code: e51b0048 e3a09000 ebfff171 eaffffa5 (e7f001f2)

cjsimon2, 2021-04-13:
@jaysuk tried your other recommendation too, got the following error:
Linux duet3 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Apr 13 10:02:52 2021 from 192.168.50.12

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@duet3:~ $ sudo mv /var/lib/dpkg/info/duet* /tmp/
pi@duet3:~ $ sudo mv /var/lib/dpkg/info/reprapfirmware.* /tmp/
pi@duet3:~ $ sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease     
Hit:2 https://pkg.duet3d.com unstable InRelease                     
Hit:3 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
pi@duet3:~ $ sudo dpkg --configure -a
pi@duet3:~ $ sudo apt-get install --reinstall duetcontrolserver
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 duetcontrolserver : Depends: duetruntime (= 3.3-b2) but 3.2.2 is to be installed
 duetsoftwareframework : Depends: duetcontrolserver (= 3.2.2) but 3.3-b2 is to be installed
 duettools : Depends: duetcontrolserver (= 3.2.2) but 3.3-b2 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
pi@duet3:~ $

jaysuk, 2021-04-13:
the output from the first test script looks more positive.
how are the board and SBC connected together?
What firmware version are you running?

cjsimon2, 2021-04-13:
@jaysuk Running this firmware: firmware-lpc-sbc-3.2.2_2.bin, on an SKR 1.4 Turbo. Connected to a pi4 via teaching techs GPIO instructions on the wiki page. I made a sys file and put the board.txt inside it. Here's the content of the board.txt file:
// Board Hardware configuration file for SKR v1.4
// generated by RepRapFirmware Configuration Tool (LPC Version) v3.2.1-LPC+9
// on Tue Apr 13 2021 01:41:50 GMT-0700 (Pacific Daylight Time)

//Note: Each line should be less than 120 characters.
//    : Unwanted options can be commented out or set to NoPin. Lines commented out will get default values
//    : for pins the default is NoPin.
//    : Values for Arrays need to be contained within { and }
//    : Comments can be defined with // or # (comments are not supported inside arrays)
//    : Each config entry must be all on a single line.

board = biquskr_1.4;

//LED blinks to indicate Platform is spinning or other diagnostic
//Comment out or set to NoPin if not wanted.
leds.diagnostic = NoPin;

heat.tempSensePins = {bedtemp, e0temp}; //Max of 3 entries
//heat.spiTempSensorCSPins = { }; //Max of 2 entries

																																				//TMC Smart Drivers
stepper.numSmartDrivers = 4;
//SBC Settings
sbc.TfrReadyPin = 1.31;
heat.tempSensePins = { 0.25, 0.24, 0.23 }
stepper.TmcDiagPins = {1.29, 1.28}

// External Display settings
serial.aux.rxTxPins = {0.3, 0.2};

jaysuk, 2021-04-13:
I would suggest the issue you're seeing now is a wiring issue.
can you hop on the discord channel so we can help you out more easily?
https://discord.gg/uS97Qs7
Thanks,

Jay

closed 2021-04-22 by gloomyandy