gloomyandy/RepRapFirmware #81: failed to start duet control server

gordonf130, 2021-04-13:
i am running skr 1,3 and pi 3b+ i have followed the instructions and tried a few fresh installs, everything is wired up correctly, im on the latest stable build, following in the automatic mode for the first time boot on the pi with the 3 lines code, the web ui shows it connects to the board but everything is blank. under status, it says no status, no tools etc, i have uploaded my printer congif files that were made with the tool for skr 1.3 and selected sbc. im a little lost as this is my first time trying to use rrf

gloomyandy, 2021-04-14:
Please use our discord server for support questions https://discord.gg/uS97Qs7

jaysuk, 2021-04-14:
this ones all sorted so can be closed

closed 2021-04-14 by jaysuk