gloomyandy/RepRapFirmware #82: M486: Object cancellation

billhu1996, 2021-04-15:
May M486 be enabled in LPC and STM?
M486: Object cancellation
#define TRACK_OBJECT_NAMES 1

jaysuk, 2021-04-15:
why do you think it doesn't work?
Sending M486 by itself (on an STM build anyway) gives an answer back,
showing that M486 is available.
Have you got all the relevant labels in your gcode file?

Thanks,

Jay

billhu1996, 2021-04-15:
Thank you for your reply.

Yes, I tried it and it didn't work.
I'm using SGen_L 1.0 with esp8266 and v3.2.2.
[image]
There are MESH related comments in my gcode

```
 92159: ;MESH:Tube
 92889: ;MESH:Cube
 93046: ;MESH:CalibrationCube
 93375: ;MESH:CalibrationCube(1)
 93754: ;MESH:NONMESH
 93768: ;MESH:Tube
 94496: ;MESH:Cube
 94647: ;MESH:CalibrationCube
 95004: ;MESH:CalibrationCube(1)

1135 matches across 1 files
```

And I saw the source code in **Pins_Duet3Mini.h**
```
#define TRACK_OBJECT_NAMES 1
```

And that there is no such definition in **Pins_LPC.h**.

Finally defined in **Pins.h**
```
#ifndef TRACK_OBJECT_NAMES
# define TRACK_OBJECT_NAMES 0
#endif
```

So I suspect that adding this switch might work.

Thanks,

Bill

jaysuk, 2021-04-15:
> I'm using Cura 4.8.

It works for me if manually indicating an object using M486 Sx
Will look into it further

Thanks,

Jay

gloomyandy, 2021-04-16:
Tracking object names is I think currently off in all versions of the firmware. The problem (on the LPC) is that of memory usage, with future versions already adding things like variables etc. there is already a lot of pressure on the available RAM. I'll add investigating what the impact would be of enabling this to my todo list.

gloomyandy, 2021-04-22:
I've enabled this feature in the V3.3-beta1_6 STM32 build. I will evaluate enabling the feature in the LPC build once I have a better understanding of the changes for v3.3-beta3 have on RAM usage.

closed 2021-04-22 by gloomyandy