gloomyandy/RepRapFirmware #84: BTT SKR2 tmc2209 are not recognized

Lino77, 2021-04-24:
[board.txt](https://github.com/gloomyandy/RepRapFirmware/files/6369587/board.txt)
[console.txt](https://github.com/gloomyandy/RepRapFirmware/files/6369588/console.txt)

jaysuk, 2021-04-24:
Remove stepper.powerEnablePin= PC.13 from your board.txt and try again

Thanks,

Jay

Lino77, 2021-04-24:
the same 
// Board Hardware configuration file for SKR2
// generated by RepRapFirmware Configuration Tool (LPC Version) v3.2.1-LPC+10 
// on Sat Apr 24 2021 10:23:34 GMT+0200 (Mitteleuropäische Sommerzeit)

//Note: Each line should be less than 120 characters.
//    : Unwanted options can be commented out or set to NoPin. Lines commented out will get default values
//    : for pins the default is NoPin.
//    : Values for Arrays need to be contained within { and }
//    : Comments can be defined with // or # (comments are not supported inside arrays)
//    : Each config entry must be all on a single line.

board = biquskr_2; 

//LED blinks to indicate Platform is spinning or other diagnostic 
//Comment out or set to NoPin if not wanted.
leds.diagnostic = PA.13; 

//TMC Smart Drivers
stepper.numSmartDrivers = 5					; Number of Smart Drivers
stepper.TmcDiagPins = {PC.1, PC.2, NoPin, NoPin, NoPin}
stepper.TmcUartPin = {PE.0, PD.3, PD.0, PC.6, PD.12}

heat.tempSensePins = {bedtemp, e0temp}		; //Max of 3 entries

	//ESP Settings
8266wifi.espDataReadyPin = PB.10;
8266wifi.TfrReadyPin = PB.11;
8266wifi.espResetPin = PC.14;
	//ESP RX/TX Settings
8266wifi.serialRxTxPins = { PD.9, PD.8 } ;
																																					
// External Display settings
serial.aux.rxTxPins = {PA.10, PA.9};

m122
=== Diagnostics ===
RepRapFirmware for STM32F4 based Boards version 3.3beta1_6 running on STM32F4
Used output buffers: 3 of 40 (19 max)
=== RTOS ===
Static ram: 30724
Dynamic ram: 63140 of which 24 recycled
Never used RAM 17112, free system stack 191 words
Tasks: NETWORK(ready,253) HEAT(delaying,218) MAIN(running,463) IDLE(ready,19)
Owned mutexes:
=== Platform ===
Last reset 00:01:03 ago, cause: [power on/off]
Last software reset details not available
Error status: 0x00
Aux0 errors 0,0,0
MCU temperature: min 33.1, current 33.2, max 33.2
Supply voltage: min 24.0, current 24.0, max 24.0, under voltage events: 0, over voltage events: 0, power good: yes
Driver 0: position 0, no-driver-detected
Driver 1: position 0, no-driver-detected
Driver 2: position 0, no-driver-detected
Driver 3: position 0, no-driver-detected
Driver 4: position 0, no-driver-detected
Driver 5: position 0
Driver 6: position 0
Driver 7: position 0
Driver 8: position 0
Driver 9: position 0
Driver 10: position 0
Date/time: 2021-04-24 16:25:19
Slowest loop: 2.50ms; fastest: 0.08ms
=== Storage ===
Free file entries: 10
SD card 0 detected
SD card longest read time 0.9ms, write time 0.0ms, max retries 0
=== Move ===
DMs created 83, maxWait 0ms, bed compensation in use: none, comp offset 0.000
=== DDARing ===
Scheduled moves 0, completed moves 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== Heat ===
Bed heaters = 0, chamberHeaters = -1
=== GCodes ===
Segments left: 0
Movement lock held by null
HTTP is idle in state(s) 0
File is idle in state(s) 0
USB is idle in state(s) 0
Aux is idle in state(s) 0
Trigger is idle in state(s) 0
Queue is idle in state(s) 0
LCD is idle in state(s) 0
Daemon is idle in state(s) 0
Autopause is idle in state(s) 0
Code queue is empty.
=== Network ===
Slowest loop: 20.78ms; fastest: 0.06ms
Responder states: HTTP(2) HTTP(0) HTTP(0) HTTP(0)
HTTP sessions: 1 of 8
- WiFi -
Network state is active
WiFi module is connected to access point 
Failed messages: pending 0, notready 0, noresp 0
WiFi firmware version 1.25-01S-D
WiFi MAC address c8:0e:14:ad:20:fe
WiFi Vcc 3.37, reset reason Turned on by main processor
WiFi flash size 4194304, free heap 35008
WiFi IP address 192.168.178.55
WiFi signal strength -53dBm, mode 802.11g, reconnections 0, sleep mode modem
Clock register 00004002
Socket states: 4 0 0 0 0 0 0 0

jaysuk, 2021-04-24:
can you join our discord channel please so we can troubleshoot this more
easily
https://discord.gg/uS97Qs7

Thanks,

Jay

Lino77, 2021-04-24:
okay found the error it looks like the board has killed all 5 drivers how can that be

closed 2021-04-29 by gloomyandy