gloomyandy/RepRapFirmware #88: RRF 3.3 not working on MKS Sbase 1.3

stisa, 2021-05-19:
Starting with version 3.3, the lpc version no longer works on my MKS sbase 1.3.
The latest stable 3.2.2 works correctly.

Version 3.3 seems to flash correctly, but then all the leds on the board are stuck on and the pc doesn't recognize the board (it doesn't show up in Device Manager at all).

I tried the first release of 3.3 and the latest RC, both in wifi and sbc version. None seem to boot.

Let me know if there's any more info that would help debugging this.

gloomyandy, 2021-05-20:
Hmm it looks like it does not like having zero TMC drivers. Please try adding the following line to your board.txt file (or modifying any existing line that may be setting it to 0)...
```
stepper.numSmartDrivers=1
```

stisa, 2021-05-21:
That was it, great!
Latest RC seems to work fine now.

One thing I noticed is that my extruder motor seems to run much hotter under RRF, could it be related? The extruder is rated 1A and set to 600mA (same as in smoothie), so I wonder if it is being set to some default current instead of the one I specify.

gloomyandy, 2021-05-22:
No idea on the current setting and no easy way to test it. What happens if you lower the current to a very low value does the motor run cooler or skip steps (which would indicate that the current setting is working). What about your other motors does it seem like the current setting is correct for them?

Smoothieboard is pretty unique in the way it sets current. The code that controls it has not been touched for a long time (I don't think I've ever even looked at it), I've not seen any other reports of issues, but then I suspect there are very few users of this board.

stisa, 2021-05-27:
Just gave it a try, sorry I took a while. 
Looks like none of the currents are getting set, M906 E1 or M906 E100  or M906 E500 or M906 E1000 there's no difference, I can still move without skipping any steps (also tried with a load).
Same for the other motors (those are Nema 17 2A), they run cooler than with smoothie (set 1.5A), my guess would be that everything get set to a default value (maybe around 800mA? Not sure how to check that).

gloomyandy, 2021-05-27:
Hi,
I've looked into this and from what I can see the setting of motor current for drivers by i2c (which is what the SBase uses) has not been enabled in any LPC build that supports TMC drivers. This means it has not been working in builds since we dropped support for the Ethernet and no network builds. So the last build that had this enabled was 3.1.1 as far as I can tell. Interestingly no one has reported the problem until now.

Anyway I have enabled the code again and it should be present in the 3.3RC3 build when that is released.

closed 2021-05-27 by gloomyandy

jaymcd, 2021-06-09:
Can I used the Ethernet port on the MKS SBase 1.3 instead of wifi. Would like to give RRF o go on it but don't really want to mess about with ESP8266's.

jaysuk, 2021-06-09:
if you use 3.1.1-14 yes. otherwise support was dropped.

jaymcd, 2021-06-10:
Gave it a go and it works pretty well.

Web initial start/ login seems a bit slow.

Any guidance on using the MKS 3.2 TFT?

I used your RRF config tool to set up the config.g file and current version of Duet Web Server. Seems ok so far. Haven't finished setting up and connecting everything and tried to print though.

Hoping to use for custom FLSun modded Cartesian cube.

Jay

Jay M. McDonald