gloomyandy/RepRapFirmware #89: Potential Bug - E3 RRF BLTouch not retracting after G29

AfroHorse, 2021-05-26:
I haven't yet been able to pinpoint whether it is the rc3 firmware but just rolled back to beta3 as I was experiencing sporadic issues where the BLTouch probe would not retract after a G29 then crash into the side of the bed. 

The console produced the following error whenever it happened "probe already triggered before probing move started" but the print would not pause automatically.

Board is a BTT E3 RRF v1.1 with an official BLTouch 3.1. All was working fine prior to a series of updates last week where I bumped up to RC3, updated wifi firmware and the DWC SD card files.

gloomyandy, 2021-05-27:
What versions are you actually using? There is currently no RC3 release, what version exactly were you using before and what version have you rolled back to? Do you still see the problem with whatever you are using now?

AfroHorse, 2021-05-27:
I have broadly kept up-to-date with the latest release so had upgraded to v3.3RC2+1_1 last week. Since the update, I experienced the issue in 25%-50% of the prints I attempted.

I have now rolled back to v3.3beta3_2 and from 5 prints this evening, I have experienced no issues with the BLTouch probe getting stuck while extended.

Attempts to use G4 (pause) or M280 (retract probe) commands in my start g-code immediately after the G29 did not seem to work either.

gloomyandy, 2021-05-27:
Unfortunately I do not have a printer with a bltouch so can't test this directly. However I'm not aware of any changes that are likely to have caused problems with a bltouch and have checked with several other users that have a bltouch and that are running v3.3RC2+1_1 and none of them have seen a similar problem. I suggest that you retest with 3.3RC3 when it is available and if you still see a problem post on our discord server and we can try and investigate further.

supvir452, 2021-05-29:
Hi there,

I am on 3.3RC2_1 and I connected up my official BLTouch 3.1 (I still use my 2.1) and did a few G29 and a couple of homing moves and the BLTouch would work each time without any hassles.

When exactly do you get this error ? only on G29? How far along was it when you get it happening? 
is the little pin still ok for the BLTouch and not bent causing it to not move as expected?

AfroHorse, 2021-05-30:
The precise point is at the end of the G29 when my start GCode would check temps were okay and then immediately move. I have been doing some testing and early results from ~5 prints is that adding a G4 immediately after my G29 seems to fix it. 

To confirm, when I reverted back to Beta3, I did experience one or 2 of the errors but less frequent than RC2. I am now on RC3 with the new start GCode (including G4) and no issues thus far.

To answer your final question, it has been bent a few times when I have no caught this error but I have re-straightened it each time and there seems to be no issues deploying and retracting normally.

supvir452, 2021-05-30:
Glad that the mystery is mostly solved then. That is always the best.

unless I am missing something you do not have to do G29 for each print - just make sure there is a proper height map that was created based on G29 and then simply load that for each print with G29 S1. You only need to redo it every now and then unless for some reason your bed level changes constantly but then there are ways to fix that.

mule1972, 2021-07-16:
I am experiencing the same issue after upgrading from 3.1.1 to 3.3-Beta2.
But I have also other issues since the update. Sometimes the bed heater does not turn off and the bed temperature stays at 50 degrees although the heater has been turned off via the GUI. Sometimes the whole system crashes and even the emergency button does nothing.
Going to setup the SBC via a full reinstall using the sd card image instead of an inplace upgrade now. Hopefully this helps.

gloomyandy, 2021-07-16:
@mule1972 Please use our discord server for support questions, you will get more responses there and it is easier for us to provide help to you.

closed 2021-07-16 by gloomyandy