gloomyandy/RepRapFirmware #94: Issues with DWC

BobCraig10, 2021-07-04:
I have no clue if this is in the correct place, sorry if it is not.  I have a BTT SKR E3 RRF on an ender 3 v2. I seem to have everything working, except in the DWC, height map only shows "Invalid Height Map"  I have run multiple meshes, next to the height map, in statistics, it has the info from the mesh. The only issue I can see is, where it says "Probe Area", after it it says, "n/a".  It does have "Probe points: 81 Probe area: n/a Maximum deviations: -0.011 mm RMS error: 0.032 mm. 
[image]

Secondly,  Under Status, where it shows tool positions and sensors, the Vin shows 24v,  all the time,  even if the only thing connected to the board is the USB with the USB power jumper on.

Thank you in advance for any help!

jaysuk, 2021-07-04:
You've not installed the correct version of DWC.
Install the one that matches your firmware version from here
https://github.com/Duet3D/DuetWebControl/releases

Only 2 boards support voltage monitoring, the fly-cdyv2 and the fly-e3-pro.
So an additional build is not required, all other boards have the voltage
set at 24v. In essence, just ignore it.

BobCraig10, 2021-07-04:
Thank you for the near immediate response!!!!I will update the DWC version now.  Thank you so much!!!!!!

BobCraig10, 2021-07-04:
> You've not installed the correct version of DWC. Install the one that matches your firmware version from here https://github.com/Duet3D/DuetWebControl/releases Only 2 boards support voltage [...]

[image]
That worked perfectly!!!!!! Thank you again for your help!

closed 2021-07-04 by BobCraig10