gloomyandy/RepRapFirmware #95: TMC5160 driver error

sanek88lbl, 2021-07-11:
These drivers work in Marlin 2.0.8.

M122:
<html>
<body>

--
11.07.2021, 23:43:32 | m122=== Diagnostics === RepRapFirmware for STM32F4 based Boards (biquskr_2) version 3.3_2 (2021-06-19 17:15:02) running on STM32F4 Used output buffers: 3 of 40 (11 max) === RTOS === Static ram: 22160 Dynamic ram: 68964 of which 40 recycled Never used RAM 38880, free system stack 182 words Tasks: NETWORK(ready,7.0%,269) HEAT(delaying,0.0%,265) Move(notifyWait,0.0%,361) TMC(delaying,6.4%,52) MAIN(running,83.7%,429) IDLE(ready,2.8%,30), total 100.0% Owned mutexes: WiFi(NETWORK) === Platform === Last reset 00:00:26 ago, cause: [power on/off] Last software reset details not available Error status: 0x00 Aux0 errors 0,2129730002,2750807100 Step timer max interval 0 MCU temperature: min 30.0, current 34.7, max 34.7 Supply voltage: min 24.0, current 24.0, max 24.0, under voltage events: 0, over voltage events: 0, power good: yes Heap OK, handles allocated/used 0/0, heap memory allocated/used/recyclable 0/0/0, gc cycles 0 Driver 0: position 0, 5160 temperature-shutdown! short-to-ground, standstill, reads 25686, writes 14, SG min/max not available Driver 1: position 0, 5160 temperature-shutdown! short-to-ground, standstill, reads 25686, writes 14, SG min/max not available Driver 2: position 0, 2209 standstill, reads 22, writes 11, error r/w 0/1, ifcnt 13, timeout 1145, failedOp 0x01, SG min/max not available Driver 3: position 0, 2209 standstill, reads 22, writes 12, error r/w 0/0, ifcnt 12, timeout 1144, failedOp 0x01, SG min/max not available Driver 4: position 0 Driver 5: position 0 Driver 6: position 0 Driver 7: position 0 Driver 8: position 0 Driver 9: position 0 Driver 10: position 0 Date/time: 2021-07-11 23:43:30 Slowest loop: 4.44ms; fastest: 0.08ms === Storage === Free file entries: 10 SD card 0 detected SD card longest read time 2.3ms, write time 0.0ms, max retries 0 === Move === DMs created 83, maxWait 0ms, bed compensation in use: none, comp offset 0.000 === DDARing === Scheduled moves 0, completed moves 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1 === Heat === Bed heaters = 0, chamberHeaters = -1 === GCodes === Segments left: 0 Movement lock held by null HTTP is idle in state(s) 0 File is idle in state(s) 0 USB is idle in state(s) 0 Aux is idle in state(s) 0 Trigger is idle in state(s) 0 Queue is idle in state(s) 0 LCD is idle in state(s) 0 Daemon is idle in state(s) 0 Autopause is idle in state(s) 0 Code queue is empty. === Network === Slowest loop: 8.56ms; fastest: 0.00ms Responder states: HTTP(2) HTTP(0) HTTP(0) HTTP(0) HTTP sessions: 1 of 8 - WiFi - Network state is active WiFi module is connected to access point  Failed messages: pending 0, notready 0, noresp 0 WiFi firmware version 1.25B0-01S-D WiFi MAC address 50:ff:20:18:93:d7 WiFi Vcc 3.38, reset reason Turned on by main processor WiFi flash size 4194304, free heap 34944 WiFi IP address 192.168.1.50 WiFi signal strength -59dBm, mode 802.11n, reconnections 0, sleep mode modem Clock register 00004002 Socket states: 0 0 0 0 0 0 0 0

</body>
</html>

[sys.zip](https://github.com/gloomyandy/RepRapFirmware/files/6797642/sys.zip)

jaysuk, 2021-07-11:
See the updated wiki info.
https://teamgloomy.github.io/skr_2.0_connected_wifi.html#tmc5160-spi-drivers
Thanks,

Jay

sanek88lbl, 2021-07-12:
> See the updated wiki info. https://teamgloomy.github.io/skr_2.0_connected_wifi.html#tmc5160-spi-drivers Thanks, Jay
> […](#)

I'm sorry, but I did it. View the attached archive -> board.txt. In slots 0 and 1 are TMC5160, in slots 2 and 3-TMC2209(2226)
Here's how it looks live:

[image]

jaysuk, 2021-07-12:
remove the "T" from the first 2 M569's in your config file

Thanks,

Jay

sanek88lbl, 2021-07-12:
> remove the "T" from the first 2 M569's in your config file Thanks, Jay

Nothing has changed(

gloomyandy, 2021-07-14:
@sanek88lbl Please use our discord server for support. It is much easier to handle discussions over there.

closed 2021-07-16 by sanek88lbl