gloomyandy/RepRapFirmware #105: LPC build + web server versions misaligned

kai1-AAA, 2021-11-06:
Hi,

lpc build 3.3.0_10 with 1.26.05 Web server shows in console - 
READ: bad format version f8 3e
READ: byte 0 val f8
READ: byte 1 val 10
READ: byte 2 val 00
READ: byte 3 val 03
READ: byte 4 val be
READ: byte 5 val fa
READ: byte 6 val b7
READ: byte 7 val 7b
READ: byte 8 val fb
READ: byte 9 val ff
READ: byte 10 val ff
READ: byte 11 val 00
READ: byte 12 val 02
READ: byte 13 val 01
READ: byte 14 val 00
READ: byte 15 val 00
READ: bad format version2 f8 3e

and drop wifi down

m122 - 
READ: === Diagnostics ===
READ: RepRapFirmware for LPC176x based Boards (biquskr_1.3) version 3.3.0_10 (2021-10-17 17:01:30) running on LPC176x at 100Mhz
READ: Used output buffers: 1 of 16 (10 max)
READ: === RTOS ===
READ: Static ram: 5604
READ: Dynamic Memory (RTOS Heap 5): 6296 free, 6296 never used
READ: Allocations: 158 Frees: 1
READ: Never used RAM 256, free system stack 196 words
READ: Tasks: NETWORK(ready,5.3%,290) HEAT(delaying,0.1%,285) Move(notifyWait,0.1%,286) MAIN(running,94.6%,311) IDLE(ready,0.0%,10), total 100.0%
READ: Owned mutexes: USB(MAIN)
READ: === Platform ===
READ: Last reset 00:07:11 ago, cause: [power up][reset button]
READ: Last software reset at 2021-11-05 23:03, reason: User, GCodes spinning, available RAM 256, slot 1
READ: Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x00400000 BFAR 0xe000ed38 SP 0x00000000 Task MAIN Freestk 0 n/a
READ: Error status: 0x00
READ: Step timer max interval 0
READ: Supply voltage: under voltage events: 0
READ: Heap OK, handles allocated/used 0/0, heap memory allocated/used/recyclable 0/0/0, gc cycles 0
READ: Driver 0: position 0
READ: Driver 1: position 0
READ: Driver 2: position 0
READ: Driver 3: position 0
READ: Driver 4: position 0
READ: Driver 5: position 0
READ: Driver 6: position 0
READ: Date/time: 1970-01-01 00:00:00
READ: Slowest loop: 2.86ms; fastest: 0.13ms
READ: Step timer: target 0 count 431690652 delta -431690653 late 0
READ: USBSerial connected 1
READ: ADC not ready 0 ADC error threshold 10 ADC Init 1
READ: === Storage ===
READ: Free file entries: 6
READ: SD card 0 detected
READ: SD card longest read time 0.9ms, write time 0.0ms, max retries 0
READ: === Move ===
READ: DMs created 40, maxWait 0ms, bed compensation in use: none, comp offset 0.000
READ: === DDARing ===
READ: Scheduled moves 0, completed moves 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
READ: === Heat ===
READ: Bed heaters = 0, chamberHeaters = -1
READ: Heater 1 is on, I-accum = 0.0
READ: === GCodes ===
READ: Segments left: 0
READ: Movement lock held by null
READ: HTTP is idle in state(s) 0
READ: File is idle in state(s) 0
READ: USB is ready with "m122" in state(s) 0
READ: Aux is idle in state(s) 0
READ: Trigger is idle in state(s) 0
READ: Queue is idle in state(s) 0
READ: Daemon is idle in state(s) 0
READ: Autopause is idle in state(s) 0
READ: Code queue is empty.
READ: === Network ===
READ: Slowest loop: 0.29ms; fastest: 0.00ms
READ: Responder states: HTTP(0) HTTP(0)
READ: HTTP sessions: 0 of 2
READ: Uploads/Errors: 0/0
READ: - WiFi -
READ: Network state is active
READ: WiFi module is idle
READ: Failed messages: pending 0, notready 0, noresp 0
READ: Bad header: 0/0
READ: WiFi firmware version 1.26-05LE-D
READ: WiFi MAC address 48:3f:da:59:f3:1e
READ: WiFi Vcc 2.94, reset reason Turned on by main processor
READ: WiFi flash size 4194304, free heap 41208
READ: Clock register 00181001
READ: Socket states: 0 5

jaysuk, 2021-11-06:
join the discord server for support

closed 2021-11-06 by jaysuk

HStakhiv, 2021-11-18:
I have the same problem. How you solved this problem?