gloomyandy/RepRapFirmware #110: Extrude not working

sankafola, 2021-12-25:
i have SKR PRO , when i do Extrude the printer do retract

sankafola, 2021-12-25:
to make sure its true i put back ver 6.1 ([gloomyandy/RepRapFirmware] New release published: v3.4.0beta6_1 ) and etrude working fine

gloomyandy, 2021-12-25:
Please use our discord server for support, it's much easier for us to handle issues over there: https://teamgloomy.github.io/support.html

gloomyandy, 2021-12-26:
This can be worked around by adding the following line to your board.txt file:
```
heat.spiTempSensorCSPins = {NoPin, NoPin, NoPin, NoPin, NoPin, NoPin, NoPin, NoPin}
```
It will be fixed in the next (post 3.4beata7_2 release).

closed 2021-12-26 by gloomyandy

sankafola, 2021-12-26:
thank you for your fast reply

sankafola, 2021-12-26:
thanks for your support