gloomyandy/RepRapFirmware #116: layer shifting

sinole1987, 2022-06-14:
Hi 

I have SKR2 with TMC2209. I have a layer shifting problem that I cannot fix no matter what setting I use. It happens in very conservative speeds, jerk and accelerations. the layer shift happens mostly on y axis where I have a 0.9 degree motor. I also have inconsistent extrusions, I also have 0.9 degree motor there. I am thinking it has something to do with 0.9 degree motors . can you help me fix this issue?

Life0fBrian, 2022-06-15:
As this does not seem to be related to RepRapFirmware this is the wrong place here for your request.

Check your printers mechanical system for easy movement, lose pulleys and properly set up the motor current and aim for proper cooling of the respective motor drivers.
Torque of 0.9° motors is often lower than with 1.8° ones! So do not use excessive speeds and accelerations.

sinole1987, 2022-06-15:
I have this problem on reprap firmware only ! my hardware is solid and I checked with marlin since you can install it on skr 2 and it works fine. either I have a wrong config in reprap or there is a bug in firmware. I should mention there is a thread on Duet forum with a bunch of people having the same problem.
I really want to run reprap on my board but this Issue has been really hard to deal with, iam currently running marlin and printing stuff everyday, so it cant be hardware. and as I mentioned I  was testing with very conservative speeds and accelerations.

sinole1987, 2022-06-15:
> If you can join us on the discord channel that would be great. We use a different 2209 implementation than the duet version of RRF and you're the only one reporting any issues. We're going to need a lot more information before we can help you
> […](#)

Hi jay 
we had an argument over fan and buck convertor, I think I annoyed you too much and you banned me on discord, I really appreciate if you can unban me so we can discuss this. I really didn't mean to bother you that night I was frustrated working on new firmware. I really don't know what I am doing wrong on firmware, I red everything regarding tuning on duetwiki and forum and everything of teamgloomy page. tried many settings and I am still having this layer shift issue.

closed 2022-06-15 by gloomyandy

jaysuk, 2022-10-11:
Done