gloomyandy/RepRapFirmware #119: [Bug]:

Oli4be, 2023-04-10:
### Duet Forum Discussion Thread

https://forum.duet3d.com/categories

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [X] Other
- [X] None

### Firmware Version

v3.4.5+_103

### Duet Web Control Version

3.4.5

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
```

### Please upload the content of your config.g file.

```gcode
; General preferences
G90                                             ; send absolute coordinates...
M83                                             ; ...but relative extruder moves
M550 P"CNC_LR3"                                 ; set printer name

; Network
M552 S0                                         ; disable network

; Drives
M569 P0 S1                                      ; physical drive 0 goes forwards using default driver timings
M569 P1 S0                                      ; physical drive 1 goes backwards using TMC2209 driver timings
M569 P2 S1                                      ; physical drive 2 goes forwards using default driver timings
M569 P3 S1                                      ; physical drive 3 goes forwards using default driver timings
M569 P4 S0                                      ; physical drive 4 goes backwards using TMC2209 driver timings
M584 X0 Y1:3 Z2:4                               ; set drive mapping // MODIFIED X on drive 0, Y on 1 and 3, Z on 2 and 4
M350 X16 Y16 Z16 E16:16 I1                      ; configure microstepping with interpolation
M92 X80.00 Y80.00 Z400.00 E80.00:80.00          ; set steps per mm
M566 X900.00 Y900.00 Z60.00 E900.00:900.00      ; set maximum instantaneous speed changes (mm/min)
M203 X3000.00 Y3000.00 Z900.00 E3000.00:3000.00 ; set maximum speeds (mm/min)
M201 X180.00 Y180.00 Z80.00 E180.00:180.00      ; set accelerations (mm/s^2)
M906 X900 Y900 Z900 E900:900 I30                ; set motor currents (mA) and motor idle factor in per cent
M84 S60                                         ; Set idle timeout

; Axis Limits
M208 X0 Y0 Z0 S1                                ; set axis minima
M208 X646 Y2092 Z129 S0                         ; set axis maxima

; Endstops
M574 X1 S1 P"^xstop"                            ; configure switch-type (e.g. microswitch) endstop for low end on X via pin xstop
M574 Y1 S1 P"^ystop+^e1stop"                    ; configure switch-type (e.g. microswitch) endstop for low end on Y via pin ystop & e1stop
;M574 Z2 S1 P"^zstop+^e0stop"                   ; configure switch-type (e.g. microswitch) endstop for high end on Z via pin zstop    
M574 Z2 S1 P"^e2stop+^e0stop" 					; configure switch-type (e.g. microswitch) endstop for high end on Z via pin e2stop & e0stop

; Z-Probe
M558 P5 C"!^zstop" H5 F120 T600                 ; set Z probe type to switch and the dive height + speeds
M558 H30                                        ; *** Remove this line after delta calibration has been done and new delta parameters have been saved
G31 P500 X0 Y0 Z14                              ; set Z probe trigger value, offset and trigger height
M557 X100:600 Y150:2000 S100                    ; define mesh grid

; Heaters
M140 H-1                                        ; disable heated bed (overrides default heater mapping)

; Fans
M950 F0 C"fan0" Q500                            ; create fan 0 on pin fan0 and set its frequency
M106 P0 S1 H-1                                  ; set fan 0 value. Thermostatic control is turned off
M950 F1 C"fan2" Q500                            ; create fan 1 on pin fan2 and set its frequency
M106 P1 S1 H-1                                  ; set fan 1 value. Thermostatic control is turned off

; Tools

; TFT
;M575 P1 S2 B250000								; Set TFT serial port & baudrate to TFT settings
M575 P1 S2 B57600 								; Alternative
;M575 P1 S0 B115200								; Alternative
												; https://teamgloomy.github.io/skr_pro_screen.html
; Custom settings
M453											; set to CNC mode
												; https://docs.duet3d.com/User_manual/Reference/Gcodes#m453-select-cnc-device-mode
```

### Please upload the content of any other releveant macro files.

```gcode
board.txt

board = biquskrpro_1.1; 

//LED blinks to indicate Platform is spinning or other diagnostic 
//Comment out or set to NoPin if not wanted.
leds.diagnostic = A.7; 

heat.tempSensePins = {}; //Max of 3 entries
//heat.spiTempSensorCSPins = { }; //Max of 2 entries

//TMC Smart Drivers
stepper.numSmartDrivers = 5;

//Connecting a screen to a SKR Pro v1.1 and v1.2
//https://teamgloomy.github.io/skr_pro_screen.html
serial.aux.rxTxPins = { PA_10, PA_9 }
```

### Details specific to your printer.

SKR Pro 1.2
BIGTREE_TFT35_V3.0_E3

### Links to additional info.

https://forum.v1e.com/t/rep-rap-firmware-on-skr-pro-1-1-or-1-2/37606/29

### What happened?

SKR Pro 1.2 running firmware-stm32f4-wifi-3.4.5+_103
BIGTREE TFT35_V3.0_E3 running BIGTREE_TFT35_V3.0_E3.27.x

Hi, 

trying to set up RRF for a CNC. The device boots with error "No Printer Attached".
However I can home all axes, and can move axes after homing. Printing/cutting is not possible due to that NPA error.
When running Marlin instead of RepRap the setup runs as expected, meaning it will be caused due to software setting/bug.

- flashed RRF v3.4.5+_103 & [v3.5beta2+_106 without succes
- flashed TFT35 FW 26 & 27 without success
- made custom config.ini & configrrf.ini without success
- rewired TFT to second serial port on SKR Pro 1.2 (& configured for it) without success
- rewired SKR to third serial port on TFT (& recompiled + configured for it) without success
- M503 returns SKR RRF config on TFT terminal correctly
- Device can home & travel, just not print due to error "No Printer Attached"
- Changed TFT serial port to all combinations, got these 3 running (with same NPA error):
M575 P1 S2 B250000	M575 P1 S2 B57600 	M575 P1 S0 B115200

All potential hardware issues have been checked, and seems not to be there.
The M503 reports back a bed temperature setting but no tool temperature. Another user on the provided url suspects the issue could be caused  in “extruders=0” case, as the CNC doesn´t use temperature it can´t be configured. That user has same hardware configured on a 3D printer and runs fine.

Can one help me troubleshoot where things go wrong in the firmware, so I can correct it to run a CNC machine?

thanks

jaysuk, 2023-04-10:
please seek support on discord, not here.
https://discord.gg/uS97Qs7

closing the issue

jaysuk, 2023-04-10:
please seek support on discord.
https://discord.gg/uS97Qs7

Thanks,

Jay

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
10/04/23,
20:55:00

closed 2023-04-10 by jaysuk